zephyrproject-rtos/zephyr #16032: Socket UDP: Low transmission efficiency

Jeepgoing, 2019-05-09:
when i upgrade zephyr from v1.13 to v1.14 i found udp send data become low efficiency.the same data v1.13(not socket) use 93us, and v1.14(socket) use 200us. who can give me some advises.

jukkar, 2019-05-09:
If you have UDP sending using native API and then in newer version use BSD socket API, you will get different results. The reason is that the socket API is built on top of native networking API (net_context). The difference looks quite big thou, I would have imagined the difference being smaller.

tbursztyka, 2019-05-09:
The summary is a bit scarce on details: which target? what's the use-case? How has this been measured?

Jeepgoing, 2019-05-09:
@jukkar , yes, i have use net_app API in 1.13, following [udp client](https://github.com/zephyrproject-rtos/zephyr/blob/v1.13-branch/samples/net/echo_client/src/udp.c)

Jeepgoing, 2019-05-09:
@tbursztyka 
1.target board is based on STM32F407
2.udp client send data to PC which from RF(radio frequency)
3.measured by
```
/* capture initial time stamp */
start_time = k_cycle_get_32();
/* do work for some (short) period of time */
...

/* capture final time stamp */
stop_time = k_cycle_get_32();
```

jukkar, 2019-05-09:
Could be also related to echo-client application which was rewritten using sockets for 1.14.

Jeepgoing, 2019-05-10:
@jukkar  yeap, i have followed echo-client sample to edit my code.

Jeepgoing, 2019-05-10:
can i use [https://github.com/zephyrproject-rtos/zephyr/blob/v1.14-branch/tests/net/udp/src/main.c](https://github.com/zephyrproject-rtos/zephyr/blob/v1.14-branch/tests/net/udp/src/main.c)?

jukkar, 2019-05-10:
> can i use https://github.com/zephyrproject-rtos/zephyr/blob/v1.14-branch/tests/net/udp/src/main.c?

You are of course free to use these lower APIs if you really want but we would like to people to use BSD socket interface as that will be a stable API for apps. Also please note that those low level APIs cannot be used from user mode as only BSD socket interface has syscall interface and memory protection implemented. 
If you have offloading enabled then the very low level APIs will not utilize that, but I suppose this is not the case here if you seek maximum performance.

Jeepgoing, 2019-05-10:
@jukkar i found if disabled gPTP, the send time from 200us to 170us. i want to know weather anyone have test the efficiency about socket udp.

jukkar, 2019-05-10:
We had earlier throughput sample app that could be used to measure some throughput values but it was removed when we removed net_app API. I have been planning to convert it to use sockets but so far I have had no time to do it.
There is also zperf network sample that can give you network timing values and throughput information.

closed 2019-09-12 by Jeepgoing