zephyrproject-rtos/zephyr #16034: Net packet size of 64 bytes doesn't work.

alexanderwachter, 2019-05-09:
**Describe the bug**
net_pkt size of 64 byte (default is 128 byte) does not work for IPv6.
**Setup**
Two nucleo F746ZG boards connected with Ethernet.
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nucleo_f746zg zephyr/samples/net/zperf
2. west build -t menuconfig. Got no Networking -> PI stack and change `Size of each network data fragment` to 64 and eactivate IPv4
3. change CONFIG_NET_CONFIG_SETTINGS=n in prj.conf
4. west flash to two boards.
5. On the first board type `zperf udp download 5001` to start the server
6. On the same board type `net iface` and copy the IPv6 address
7. On the other board type `zperf udp upload <IP from clipboard> 5001 10 1K 10M`

**Expected behavior**
Don't show `LAST PACKET NOT RECEIVED!!!`

**Impact**
Optimization for L2 with small MTU not possible.

**Screenshots or console output**
***Client***
```
uart:~$ zperf udp upload fe80::280:e1ff:fe34:129a 5001 10 1K 10M

Remote port is 5001
Connecting to fe80::280:e1ff:fe34:129a
Duration:       10.00 s
Packet size:    1024 bytes
Rate:           10240 kbps
Starting...
Rate:           10.00 Mbps
nb_packets=1081 delay=0 adjust=-9944
nb_packets=2157 delay=0 adjust=-10631
nb_packets=3233 delay=0 adjust=-10321
nb_packets=4309 delay=0 adjust=-10154
nb_packets=5381 delay=0 adjust=-10583
nb_packets=6458 delay=0 adjust=-10297
nb_packets=7532 delay=0 adjust=-10422
nb_packets=8608 delay=0 adjust=-10271
nb_packets=9684 delay=0 adjust=-10571
-
Upload completed!
LAST PACKET NOT RECEIVED!!!
Statistics:             server  (client)
Duration:               0 us    (10.00 s)
Num packets:            0       (10757)
Num packets out order:  0
Num packets lost:       0
Jitter:                 0 us
Rate:                   0 Kbps  (8.40 Mbps)
```
***Server***
```
uart:~$ zperf udp download 5001

Binding to ::
Listening on port 5001
UDP server started on port 5001
[00:01:36.423,000] <err> net_pkt: Uncontiguous data cannot be linearized
[00:01:36.423,000] <err> net_ipv6: DROP: NULL NS header
[00:01:37.431,000] <err> net_pkt: Uncontiguous data cannot be linearized
[00:01:37.431,000] <err> net_ipv6: DROP: NULL NS header
[00:01:38.677,000] <err> net_pkt: Uncontiguous data cannot be linearized
[00:01:38.677,000] <err> net_ipv6: DROP: NULL NS header
[00:01:39.683,000] <err> net_pkt: Uncontiguous data cannot be linearized
[00:01:39.683,000] <err> net_ipv6: DROP: NULL NS header
[00:01:40.925,000] <err> net_pkt: Uncontiguous data cannot be linearized
[00:01:40.925,000] <err> net_ipv6: DROP: NULL NS header
[00:01:42.166,000] <err> net_pkt: Uncontiguous data cannot be linearized
[00:01:42.166,000] <err> net_ipv6: DROP: NULL NS header
[00:01:43.636,000] <err> net_pkt: Uncontiguous data cannot be linearized
[00:01:43.636,000] <err> net_ipv6: DROP: NULL NS header
[00:01:44.869,000] <err> net_pkt: Uncontiguous data cannot be linearized
[00:01:44.869,000] <err> net_ipv6: DROP: NULL NS header
[00:01:45.876,000] <err> net_pkt: Uncontiguous data cannot be linearized
[00:01:45.876,000] <err> net_ipv6: DROP: NULL NS header
[00:01:46.884,000] <err> net_pkt: Uncontiguous data cannot be linearized
[00:01:46.884,000] <err> net_ipv6: DROP: NULL NS header
[00:01:48.128,000] <err> net_pkt: Uncontiguous data cannot be linearized
[00:01:48.128,000] <err> net_ipv6: DROP: NULL NS header
[00:01:51.352,000] <err> net_pkt: Uncontiguous data cannot be linearized
[00:01:51.352,000] <err> net_ipv6: DROP: NULL NS header
[00:07:23.649,000] <err> net_pkt: Uncontiguous data cannot be linearized
[00:07:23.649,000] <err> net_ipv6: DROP: NULL NS header
[00:07:24.890,000] <err> net_pkt: Uncontiguous data cannot be linearized
[00:07:24.890,000] <err> net_ipv6: DROP: NULL NS header
[00:07:26.134,000] <err> net_pkt: Uncontiguous data cannot be linearized
[00:07:26.134,000] <err> net_ipv6: DROP: NULL NS header
```

alexanderwachter, 2019-05-09:
@jukkar I'll look into that but with low priority because it's not very important to support 64 byte net_pkt size.

tbursztyka, 2019-05-09:
@alexanderwachter check ipv6_nbr.c, in function handle_ns_input(), just change NET_PKT_DATA_ACCESS_CONTIGUOUS_DEFINE(ns_access, ...
to:
NET_PKT_DATA_ACCESS_DEFINE(ns_access, ...

It will raise the stack usage in that function by a sizeof(struct net_icmpv6_ns_hdr) but that's ok.

Check the other input handlers, looks like they all have the same "issue".

Note however that IP headers themselves are assumed to be always in a contiguous area and that should stick like that. (extended headers, and so on, it's alright not to be in a contiguous area).

alexanderwachter, 2019-05-09:
Thanks @tbursztyka I'll look at it tomorrow. IP headers have a fixed size of 40 bytes. Thats why i chose 64. I don't want to go lower than that.

jukkar, 2020-01-28:
Going through old issues and this is probably not going to be worked on any time soon. Can be re-opened if really needed.

closed 2020-01-28 by jukkar