zephyrproject-rtos/zephyr #16035: facing problem with SDHC driver disk mount, need help to debug better

Pathak94, 2019-05-09:
I'm getting these error with SDHC driver, the 'Faulting instruction address' doesn't exist. I tried debugging by enabling ASSERTS, stack  but couldn't find the reason behind this. I think this error is originating from fs_mount but don't know what is the issue.

``` 
******* BUS FAULT *****
  Instruction bus error
***** Hardware exception *****
Current thread ID = 0x20000018
Faulting instruction address = 0x101977a
Fatal fault in thread 0x20000018! Aborting.**
```

This is the fault that seems to happen most of the time, but I have seen precise bus fault also (faulting instruction in `find_volume`), and MPU_FAULT.

Following is the code snippet, it is a thread apart from the main thread with priority 5.
```
void logging_fn()
{
     /* initialise SD card and file system*/
	mp.mnt_point = disk_mount_pt;
	k_sleep(5000);
	int res = fs_mount(&mp);
	if (res == FR_OK) {
		LOG_INF("Disk mounted.\n");
	}
	else {
		LOG_ERR("Error mounting disk.\n");
	}
	struct fs_file_t file;
	int total_samples = 0;
	char output_buffer[100];
	int rc = fs_open(&file, "/SD:/myfile.csv");
	if (rc != 0) {
		LOG_ERR("file not created\n");
	}
	while (1) {	
		k_msgq_get(&logging_q, &val, K_FOREVER);	
		if (total_samples < 10) {
		    int written = snprintf(output_buffer, sizeof(output_buffer)-1, "%10.2f, %10.2f,          %10.2f, %10.2f, %10.2f, %10.2f\n",
			sensor_value_to_double(&val.accel[0]),
			sensor_value_to_double(&val.accel[1]),
			sensor_value_to_double(&val.accel[2]),
			sensor_value_to_double(&val.brightness),
			sensor_value_to_double(&val.humi),
			sensor_value_to_double(&val.temp));	    
			output_buffer[sizeof(output_buffer)-1] = 0;
			int w_size = fs_write(&file, output_buffer, written);
		    if (w_size > 0) {
			    LOG_INF("%d bytes written\r\n", w_size);
		    }
		    else {
			    LOG_INF("err in writing");
		    }
            total_samples++;
		}
		if(total_samples == 10) {
			if (fs_close(&file) != 0) {
			    LOG_ERR("file not closed");
		    }
			k_sleep(1500);
	        if(fs_unmount(&mp) != 0) {
				LOG_ERR("Unable to unmount");
			}
			k_sleep(1500);
		total_samples++;
		}
	}
}
```

The above code functions like this 

1. mounts the dis "SD:" as fat file system.
3. gets the message from the message queue.
4. fills the buffer and writes 10 samples to the file.
5. close the file and unmount after 10 samples.

Also, there is a 5000ms delay at the start because otherwise mounting gives random exceptions, and not consistently either.

This is the overlay file.
```
&spi1 {
        status = "ok";
        cs-gpios = <&gpio0 17 0>;

        sdhc0: sdhc@0 {
                compatible = "zephyr,mmc-spi-slot";
                reg = <0>;
                status = "ok";
                label = "SDHC0";
                spi-max-frequency = <24000000>;
        };
};
```

I'm currently stuck at this point. I need help to debug this better.

nvlsianpu, 2019-05-13:
Does sample /samples/subsys/fs/src/main.c works for you?
What a board you are using?

Pathak94, 2019-05-13:
It used to, but recently I started getting this error.
```
***** Booting Zephyr OS zephyr-v1.14.0 *****
[00:00:00.002,990] <inf> spi_nrfx_spi: CS control inhibited (no GPIO device)
Error mounting disk.
[00:00:00.032,165] <err> main: Storage init ERROR!
[00:00:00.062,103] <err> fs: fs mount error (-5) 
```

Pathak94, 2019-05-13:
I'm using nrf52840_blip board.

nvlsianpu, 2019-05-13:
```
spi_nrfx_spi: CS control inhibited (no GPIO device)
```
@anangl - is this a regresion?

ntavish, 2019-05-13:
I was using same hardware when I added the SDHC sample in #12783. It was working then.

ntavish, 2019-05-13:
Also found that the clock speed for SPI is always about 250KHz; even though according to the code in `sdhc_detect(...)` it should start at 400KHz, then after initialization, it should switch to 4MHz.

[image]

mrfuchs, 2019-07-02:
I experienced similar issues using my custom STM32F4 board. While `fs_mount()` will always work on its first call, it will almost always fail with `FR_DISK_ERR` (error -5, that is) on subsequent calls after `fs_unmount()` has been called in between.
It looks like a SDHC disk - once it was initialized - never gets re-initialized, no matter if a failure occured afterwards or it even was unmounted, ejected, replaced and re-mounted.

The easy way to get around this issue is to unconditionally initialize the SDHC card:

```diff
diff --git a/subsys/disk/disk_access_sdhc.c b/subsys/disk/disk_access_sdhc.c
index 7460c0623d..15ac52e29e 100644
--- a/subsys/disk/disk_access_sdhc.c
+++ b/subsys/disk/disk_access_sdhc.c
@@ -994,10 +999,10 @@ static int disk_sdhc_access_init(struct disk_info *disk)
 	struct sdhc_data *data = dev->driver_data;
 	int err;
 
-	if (data->status == DISK_STATUS_OK) {
+//	if (data->status == DISK_STATUS_OK) {
 		/* Called twice, don't re-init. */
-		return 0;
-	}
+//		return 0;
+//	}
 
 	err = sdhc_detect(data);
 	sdhc_set_cs(data, 1);
```

nvlsianpu, 2019-08-13:
@ntavish Can you propose a fix for this bug?

ntavish, 2019-08-16:
@nvlsianpu sorry, I've not been able to look at this right now. I'll see if I can do something next week.

nvlsianpu, 2019-09-02:
@ntavish Any progress on that?

dleach02, 2019-10-15:
@ntavish, does your #19391 PR supposed to address this bug?

ntavish, 2019-10-15:
@dleach02 I think it should, I am not going to be able to test it thoroughly for a while though, so if possible please test. In whatever tests with the fs sample I did, and also according to @mrfuchs, it should fix this issue.

ioannisg, 2019-11-21:
@anangl @nvlsianpu can you help with this ticket? It's effectively blocking the 2.1 release - I wonder if we have a way to address this now. MAybe look at #19391? @carlescufi I would remove the Medium priority labeling on this one if we can't resolve it soon.

nvlsianpu, 2019-11-22:
Both me and @anangl can't help with the ticket right now (even don't have appropriate HW around my desk).

barsok, 2019-11-22:
> I don't understand why this issue is labeled platform: nrf.

Neither do I.   "CS inhibited" is the result of a wrong (or missing) gpio configuration  that is external to the SPI driver. I would opt for removing "platform: nrf" from the ticket and checking the context given to transceive() function call.

dleach02, 2019-11-26:
I'm not sure why I added the nrf platform. May have been a fat finger but @ntavish has a PR that he believes will fix this.

PR #19391 (2019-11-27): disk: sdhc: reinitialize and detect always
Do not skip sdhc/spi sdhc disk detect and interface init, on subsequent calls. This pr does not touch flash and RAM disk files as those will not need 'detection' of storage medium. It also does not touch USDHC initialization.

See https://github.com/zephyrproject-rtos/zephyr/issues/16035#issuecomment-507651886.

Fixes #16035
closed 2019-11-27 by MaureenHelm