zephyrproject-rtos/zephyr #16041: stm32f407 flash erase error sometimes

maweicoltsmart, 2019-05-09:
int flash_stm32_wait_flash_idle(struct device *dev)
{
	s64_t timeout_time = k_uptime_get() + STM32_FLASH_TIMEOUT;
	int rc;

	rc = flash_stm32_check_status(dev);
	if (rc < 0) {
		printk("%s, %d\r\n",__func__,__LINE__);
		return -EIO;
	}

	while ((FLASH_STM32_REGS(dev)->sr & FLASH_SR_BSY)) {
		if (k_uptime_get() > timeout_time) {
			return -EIO;
		}
	}

	return 0;
}
stm32f407vet6 flash erase error sometimes，program run to line printk above

maweicoltsmart, 2019-05-09:
I erase address 0x8060000

erwango, 2019-07-03:
@maweicoltsmart do you still face this issue?
If yes, can you try to increase STM32_FLASH_TIMEOUT?

erwango, 2019-07-26:
@maweicoltsmart , we've increased STM32_FLASH_TIMEOUT since it was seen to be too tight in some cases. The change is available in current master.
Can you try to see if you still see the issue with this change?

erwango, 2019-11-01:
@maweicoltsmart, is this issue still valid?

erwango, 2019-11-21:
No update, can't be reproduced. Likely fixed by STM32_FLASH_TIMEOUT increased flashed some months ago.
I'm closing the point. Please feel free to re-open if seen again

closed 2019-11-21 by erwango