zephyrproject-rtos/zephyr #16046: modules are being processed too late.

nashif, 2019-05-09:
Not able to use mbedTLS as a module due to missing dependencies and wrong order of processing of cmake files.

```
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
CMake Error at ../../../../cmake/extensions.cmake:962 (target_link_libraries):
  Cannot specify link libraries for target "mbedTLS" which is not built by
```

SebastianBoe, 2019-05-10:
I am unable to reproduce this problem.

I tried using TINYCBOR as a module, and did not get any problems.

```
diff --git a/samples/hello_world/prj.conf b/samples/hello_world/prj.conf
index b2a4ba5910..cef1cc481c 100644
--- a/samples/hello_world/prj.conf
+++ b/samples/hello_world/prj.conf
@@ -1 +1,3 @@
 # nothing here
+CONFIG_FILE_SYSTEM=y
+CONFIG_TINYCBOR=y
diff --git a/subsys/fs/CMakeLists.txt b/subsys/fs/CMakeLists.txt
index 5f296d8ee3..d639c9758f 100644
--- a/subsys/fs/CMakeLists.txt
+++ b/subsys/fs/CMakeLists.txt
@@ -12,6 +12,7 @@ if(CONFIG_FILE_SYSTEM)
   zephyr_library_sources_ifdef(CONFIG_FILE_SYSTEM_SHELL  shell.c)
 
   zephyr_library_link_libraries(FS)
+  zephyr_library_link_libraries(TINYCBOR)
 
   target_link_libraries_ifdef(CONFIG_FAT_FILESYSTEM_ELM FS INTERFACE ELMFAT)
   target_link_libraries_ifdef(CONFIG_FILE_SYSTEM_NFFS   FS INTERFACE NFFS)

```

nashif, 2019-05-10:
try mbedtls or fat filesystem. The issue is there.

SebastianBoe, 2019-05-10:
Which library links with which library, and which library is a module when this can be reproduced?

SebastianBoe, 2019-05-10:
Or just dump the code.

SebastianBoe, 2019-05-10:
Still not able to reproduce, now with mbedtls as a module. And the fat filesystem linking with mbedtls.

nashif, 2019-05-10:
just run sanitycheck:

```
i9:zephyr(modularize_master_mbedtls): sanitycheck
JOBS: 40
Cleaning output directory /home/nashif/Work/split/zephyr/sanity-out
Selecting default platforms per test case
Building testcase defconfigs...
Could not build defconfig for frdm_k64f_samples_net_sockets_big_http_download_sample.net.sockets.big_http_download_config-sanitycheck
	see: /home/nashif/Work/split/zephyr/sanity-out/frdm_k64f/samples/net/sockets/big_http_download/sample.net.sockets.big_http_download/config-sanitycheck.log
Could not build defconfig for mps2_an385_samples_net_sockets_big_http_download_sample.net.sockets.big_http_download_config-sanitycheck
	see: /home/nashif/Work/split/zephyr/sanity-out/mps2_an385/samples/net/sockets/big_http_download/sample.net.sockets.big_http_download/config-sanitycheck.log
```

nashif, 2019-05-10:
```
Zephyr version: 1.14.99
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.7.2", minimum required is "3.4")
-- Selected BOARD frdm_k64f
-- Found west: /home/nashif/.local/bin/west (found suitable version "0.5.7", minimum required is "0.5.6")
-- Loading /home/nashif/Work/split/zephyr/boards/arm/frdm_k64f/frdm_k64f.dts as base
-- Overlaying /home/nashif/Work/split/zephyr/dts/common/common.dts

warning: TEST_RANDOM_GENERATOR (defined at subsys/random/Kconfig:8) was assigned the value 'y' but
got the value 'n'. You can check symbol information (including dependencies) in the 'menuconfig'
interface (see the Application Development Primer section of the manual), or in the Kconfig
reference at
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_TEST_RANDOM_GENERATOR.html (which is
updated regularly from the master branch). See the 'Setting configuration values' section of the
Board Porting Guide as well.
Parsing Kconfig tree in /home/nashif/Work/split/zephyr/Kconfig
Loading /home/nashif/Work/split/zephyr/boards/arm/frdm_k64f/frdm_k64f_defconfig as base
Merging /home/nashif/Work/split/zephyr/samples/net/sockets/big_http_download/prj.conf
Configuration written to '/home/nashif/Work/split/zephyr/sanity-out/frdm_k64f/samples/net/sockets/big_http_download/sample.net.sockets.big_http_download/zephyr/.config'
-- Cache files will be written to: /home/nashif/.cache/zephyr
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/nashif/Work/SDK/0.10.0.73/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
CMake Error at ../../../../cmake/extensions.cmake:1011 (target_link_libraries):
  Cannot specify link libraries for target "mbedTLS" which is not built by
  this project.
Call Stack (most recent call first):
  ../../../../subsys/net/lib/sockets/CMakeLists.txt:19 (zephyr_link_interface_ifdef)

-- Configuring incomplete, errors occurred!
See also "/home/nashif/Work/split/zephyr/sanity-out/frdm_k64f/samples/net/sockets/big_http_download/sample.net.sockets.big_http_download/CMakeFiles/CMakeOutput.log".
See also "/home/nashif/Work/split/zephyr/sanity-out/frdm_k64f/samples/net/sockets/big_http_download/sample.net.sockets.big_http_download/CMakeFiles/CMakeError.log"
```

SebastianBoe, 2019-05-10:
Will use #16065 16065 to reproduce.

SebastianBoe, 2019-05-10:
Reproduced.

Found the bug (at least the bug provoked by big_http_download).

The bug is not related to the ordering of mbedTLS and I can not explain why it was only provoked now.

This is the root cause:

This line, https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/lib/sockets/CMakeLists.txt#L19 , should be removed. It is modifying mbedTLS. I don't know why, but it is not sane for sockets to do so. Not only is it insane, it is also redundant, as mbedTLS already does the same modification in it's own build scripts:

https://github.com/zephyrproject-rtos/zephyr/blob/master/ext/lib/crypto/mbedtls/CMakeLists.txt#L117

Removing that line fixes the sample.

nashif, 2019-05-10:
there are other tests failing, for example: tests/subsys/jwt

nashif, 2019-05-10:
here is a list of failing tests (with fix from above comment):
```
i9:zephyr(modularize_master_mbedtls): sanitycheck -p qemu_x86
JOBS: 40
Cleaning output directory /home/nashif/Work/split/zephyr/sanity-out
Building testcase defconfigs...
Filtering test cases...
275 tests selected, 103273 tests discarded due to filters
total complete:    0/ 272   0%  failed:    0

qemu_x86                  samples/drivers/crypto/sample.driver.crypto.mbedtls FAILED: build_error
	see: sanity-out/qemu_x86/samples/drivers/crypto/sample.driver.crypto.mbedtls/build.log

total complete:    1/ 272   0%  failed:    1

qemu_x86                  samples/net/lwm2m_client/sample.net.lwm2m_client.dtls FAILED: build_error
	see: sanity-out/qemu_x86/samples/net/lwm2m_client/sample.net.lwm2m_client.dtls/build.log

total complete:  120/ 272  44%  failed:    2

qemu_x86                  tests/crypto/mbedtls/crypto.mbedtls                FAILED: handler_crash
	see: sanity-out/qemu_x86/tests/crypto/mbedtls/crypto.mbedtls/run.log

total complete:  179/ 272  65%  failed:    3

qemu_x86                  tests/net/all/net.build                            FAILED: build_error
	see: sanity-out/qemu_x86/tests/net/all/net.build/build.log

total complete:  232/ 272  85%  failed:    4

qemu_x86                  tests/subsys/jwt/libraries.encoding                FAILED: build_error
	see: sanity-out/qemu_x86/tests/subsys/jwt/libraries.encoding/build.log
```

SebastianBoe, 2019-05-10:
I have now tested and fixed these failures here: https://github.com/zephyrproject-rtos/zephyr/pull/16069/files

nashif, 2019-06-19:
should be fix now

PR #16083 (2019-05-27): [Backport v1.14-branch] cmake: Don't have users call zephyr_link_interface on mbedTLS
Backport #16069.

Fixes #16046
closed 2019-06-19 by nashif