zephyrproject-rtos/zephyr #16052: Adafruit Trinket M0 Bossac Offset is Wrong

psidhu, 2019-05-09:
**Describe the bug**
When using bossac to flash the adafruit_trinket_m0, it defaults to the 0x0000 address. This is wrong due to the UF2 bootloader. See https://learn.adafruit.com/adafruit-trinket-m0-circuitpython-arduino/uf2-bootloader-details#running-bossac-on-the-command-line-46-29.

Also, due to https://github.com/zephyrproject-rtos/zephyr/issues/13151 (code,partition not taking effect unless BOOTALODER_MCUBOOT is set), simply fixing the bossac offset isn't enough. The physical location of where the application is flashed must change.

Note that adafruit ships these locked now: https://forums.adafruit.com/viewtopic.php?f=57&t=142170&p=707151#p707153

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=adafruit_trinket_m0
3. make
4. make flash
5. see error

**Expected behavior**
For bossac to flash to the correct address for this board.

**Impact**
Showstopper as no one can flash this board with `make flash`

ixmeins, 2019-05-11:
I used the Adafruit board as a basis for my homegrown SAMR21 board and I noticed this, as well as the defaults for my particular selected chip (now that R21 support is built in) being wrong with a default start address of 0x0000, 256kb flash and 32k ram where as my chip is 128/16. I also used a modified Adafruit bootloader so also needed the 0x2000 start.
I get around it by resetting those values (ninja menuconfig general architecture options) but then I use the UF2 method to program as I could not get bossac to response via ninja flash.

closed 2019-05-12 by nashif

sslupsky, 2019-11-25:
see #20740