zephyrproject-rtos/zephyr #16079: SAM0/SAMR SERIAL subsystem broken?

ixmeins, 2019-05-11:
Since ADC and GPIO interrupt support was added the SERIAL (UART) subsystem does not appear to work.
I can compile and run various examples (/samples/hello_world etc) in 1.14 LTS but the latest zephyr git clone seems to have broken the serial io. I get no serial output after successful compiles. I'm using a SAMR21E17A, previously with my own SAMR21 support and now with the integrated support. 
I have an application which uses the USB CDC_ACM drivers and the shell subsystem which works fine under LTS but also does not work (after a successful compile) under the latest version.
I'm a Zephyr newb but have successfully been working around the various issues like wrong compile addresses and SOC parameters not being passed correctly based on user selected chip.
Any ideas appreciated,

ixmeins, 2019-05-11:
I've just built the SHELL sample app on 1.14 LTS targeting the Adafruit_Feather_Basic_Proto (SAMD21) using CDC_ACM as the I/O instead of SERCOM and it compiled and ran as expected. I then compiled the same app using the latest version of Zephyr and again it compiles fine but the CDC_ACM USB device is not enumerated and so the application does nothing.

ixmeins, 2019-05-11:
O.K.. I'm likely confused. I just built the Adafruit_Feather..ect version of the purely serial (SERCOM0) shell example and it built and ran correctly. I thought I also tried that on the SAMR21 board without success.

Maybe I really just need confirmation from a tester that after the addition of the ADC & GPIO Interrupt code for the SAM0 (SAMD21/R21) they tested that the USB CDC_ACM builds and runs correctly on SAMR21 and SAMD21 including with the SHELL subsystem?

Sizurka, 2019-05-11:
Hmm, I was able to get `samples/subsys/shell/shell_module` working on a SAMD21 based board with:
```Kconfig
CONFIG_SHELL_BACKEND_SERIAL=y
CONFIG_USB=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_PRODUCT="Zephyr USB console sample"
CONFIG_SHELL_BACKEND_SERIAL_INTERRUPT_DRIVEN=y
CONFIG_USB_CDC_ACM=y
CONFIG_UART_SHELL_ON_DEV_NAME="CDC_ACM_0"

CONFIG_ADC=y
CONFIG_ADC_ASYNC=y
CONFIG_GPIO=y
```
With and without the last section, it acts the same.

So I'm not sure it's related to the ADC/GPIO changes.  I don't have any of the Adafruit boards or the SAMD21/R21 Xpro ones, so I can't test those specifically.  I am not familiar with the internals of the shell though, so maybe something there changed?

It looks like there might be some problems with the intersection of the shell and `CONFIG_USB_UART_CONSOLE=y` combined with `CONFIG_USB_UART_DTR_WAIT=y` (i.e. it looks like that gets into a fatal error somehow, probably while waiting for the host connection).  I'm not sure I'll have time in the near future to debug that further, but that might be a starting point, if you're trying to use both of those.

ixmeins, 2019-05-12:
Yes it is unlikely to be in the GPIO/ADC I guess. I'll have to go and see if I can find any changes that were introduced since the LTS version related to SPI or the SHELL.
Thanks for looking.

closed 2019-05-12 by ixmeins