zephyrproject-rtos/zephyr #16080: Zephyr UART shell crashes on start if main() is blocked

frantony, 2019-05-11:
@jarz-nordic @nordic-krch

**Describe the bug**
Zephyr UART shell crashes on start if main() is blocked.

**To Reproduce**
Steps to reproduce the behavior:

1. compile ```samples/subsys/shell``` with blocked ```main()```

```shell
patch -p1 <<EOF;
--- a/samples/subsys/shell/shell_module/src/main.c
+++ b/samples/subsys/shell/shell_module/src/main.c
@@ -120,4 +120,5 @@ SHELL_CMD_ARG_REGISTER(version, NULL, "Show kernel version", cmd_version, 1, 0);
 void main(void)
 {
 
+	while(1);
 }
EOF

cd samples/subsys/shell/shell_module
rm -rf build && mkdir build && cd build
cmake -DBOARD=olimexino_stm32  ..
make
```
2. burn zephyr image into MCU;
3. open your favorite communication application (e.g. minicom or screen) and connect to MCU UART_1;
5. reset MCU;
6. see no characters from MCU UART_1 (one one garbage character sometimes);
4. send one character into MCU UART_1;
6. see zephyr crash in the communication application:
```
***** BUS FAULT *****
  Imprecise data bus error
***** Hardware exception *****
Current thread ID = 0x200005f0
Faulting instruction address = 0x80056d4
Fatal fault in ISR! Spinning...
```

**Expected behavior**
I expect to see zephyr UART_1 shell prompt in my communication application just after reset.

**Impact**
showstopper. If main thread is not blocked but introduce just delay (e.g. USB CDC ACM initial activity) I see the same crash on start.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Zephyr git describe: zephyr-v1.14.0-712-g59eb48329e

bbolen, 2019-05-12:
I've been sitting on this patch for a little while, but I think it addresses the issue you are seeing.  Let me know if it helps.

From 78e8ec84f8040afef0e7b6a8caa57e5bdf156596 Mon Sep 17 00:00:00 2001                                                                                                                                      
From: Bradley Bolen <bbolen@lexmark.com>
Date: Thu, 14 Feb 2019 11:28:37 -0500
Subject: [PATCH] shell: Move the call to uart_irq_rx_enable to the enable
 callback

The uart could get an RX interrupt before the main shell init code has 
finished initializing.  The race looks like:

shell.c                   shell_uart.c
shell_init()
 ->instance_init()
                          init()
                           ->uart_irq_init()
                            ->uart_irq_callback_user_data_set()

                          RX interrupt
                          uart_rx_handle()
                           ->sh_uart->ctrl_blk->handler

transport_evt_handler()
 ->k_poll_signal_raise()

 ->k_thread_create(shell_thread)

Unfortunately, the code that initializes the signals is not run until
the beginning of shell_thread.  But that comes after
k_poll_signal_raise() has alread been called.
---
 subsys/shell/shell_uart.c | 5 ++++-
 1 file changed, 4 insertions(+), 1 deletion(-)

diff --git a/subsys/shell/shell_uart.c b/subsys/shell/shell_uart.c
index f9dba45..ce3333c 100644
--- a/subsys/shell/shell_uart.c
+++ b/subsys/shell/shell_uart.c
@@ -137,7 +137,6 @@ static void uart_irq_init(const struct shell_uart *sh_uart)
    struct device *dev = sh_uart->ctrl_blk->dev;
 
    uart_irq_callback_user_data_set(dev, uart_callback, (void *)sh_uart);
-   uart_irq_rx_enable(dev);
 #endif
 }
 
@@ -195,6 +194,10 @@ static int enable(const struct shell_transport *transport, bool blocking_tx)
 #endif
    }   
 
+#ifdef CONFIG_SHELL_BACKEND_SERIAL_INTERRUPT_DRIVEN
+   uart_irq_rx_enable(sh_uart->ctrl_blk->dev);
+#endif
+
    return 0;
 }
 
-- 
2.7.4

frantony, 2019-05-12:
@bbolen 

I have just tested your patch on top of 59eb48329e. It works perfectly and solves my problem.
Thanks!

jukkar, 2019-05-12:
@bbolen Could you send a PR for this issue as you have a working solution?

ixmeins, 2019-05-13:
fyi I've tried this patch on the latest repo and it breaks cdc_acm shell comms. If I put the rx irq enable back where it was then cdc_acm shell works again. The symptom is that outgoing messages will appear in the terminal but no incoming traffic is seen by the shell.

PR #16093 (2019-05-13): shell: Move signals initialization from thread to init
By moving signals initialization to shell instance init function,
shell instance is ready to receive RX signals from backend before
thread is ready to handle them.

Fixes #16080.
PR #21164 (2019-12-05): [Backport 16093 to v1.14 branch]shell: Move signals initialization from thread to init
By moving signals initialization to shell instance init function,
shell instance is ready to receive RX signals from backend before
thread is ready to handle them.

Fixes #16080.
closed 2019-05-13 by carlescufi