zephyrproject-rtos/zephyr #16089: Mcux Ethernet driver does not detect carrier anymore (it's alway on)

tbursztyka, 2019-05-13:
**Describe the bug**
Mcux Ethernet driver used to be able to detect carrier state, but something somewhere has changed and it things the carrier is always on now.

**To Reproduce**
Do not plug the Ethernet cable to your device
Use samples/net/sockets/echo_client

1. mkdir build; cd build
2. Change the prj.conf so it becomes:
```CONFIG_NET_CONFIG_SETTINGS=y
#CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::2"
#CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::1"
#CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.0.2.2"
#CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.0.2.1"
CONFIG_NET_DHCPV4=y
CONFIG_NET_L2_ETHERNET_MGMT=y
CONFIG_NET_MGMT_EVENT_LOG_LEVEL_DBG=y
CONFIG_NET_DHCPV4_LOG_LEVEL_DBG=y
```
3. cmake -DBOARD=board\frdm_k64f
4. make
5. Dhcp starts right away to send discover
6. Plug the Ethernet cable: no changes. Eventually the discover timeout will restart and dhcp will be successful but that's the wrong behavior: it should receive a if_up net_mgmt event then starts dhcp discover.

**Expected behavior**
Carrier on/off (and thus net if up/down state) should be detected

**Impact**
Unable to test properly code that rely on carrier off/on and net if up/down events on that device.

MaureenHelm, 2019-07-05:
@agansari can you take a look at this?

ioannisg, 2019-08-01:
@tbursztyka pls, try the patch in #17847 and report if this fixes the bug :)

tbursztyka, 2019-08-02:
I'll try to find some time next week, too busy atm.

PR #17847 (2019-08-08): drivers: eth_mcux: event corrected init->reset
Fixes #16089
closed 2019-08-08 by ioannisg