zephyrproject-rtos/zephyr #16103: nrf5 802.15.4 driver requires Log subsys to be enabled

jcorde, 2019-05-13:
**Describe the bug**
Compiling OpenThread while CONFIG_LOG=n leads to an undeclared expression error.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Hello world sample
2. Enable OT and disable Log by adding the following lines to its `prj.conf` file:
`CONFIG_LOG=n
CONFIG_NET_L2_OPENTHREAD=y
CONFIG_NETWORKING=y
CONFIG_FLASH=y
CONFIG_FLASH_PAGE_LAYOUT=y
CONFIG_CPLUSPLUS=y
CONFIG_REBOOT=y`
 
3. Build it for nrf52840:
`west -v build -p -b nrf52840_pca10059`

4. Wait for compilation failure.

**Expected behavior**
Successful compile.

**Impact**
Showstopper.

**Console output**
```
[192/200] Building C object zephyr/CMakeFiles/zephyr.dir/drivers/ieee802154/ieee802154_nrf5.c.obj
FAILED: ccache /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=zephyr-v1.14.0-695-g43492817ab37 -DKERNEL -DNRF52840_XXAA -DNRF_802154_ACK_TIMEOUT_ENABLED=0 -DNRF_802154_CCA_CORR_LIMIT_DEFAULT=2 -DNRF_802154_CCA_CORR_THRESHOLD_DEFAULT=45 -DNRF_802154_CCA_ED_THRESHOLD_DEFAULT=45 -DNRF_802154_CCA_MODE_DEFAULT=NRF_RADIO_CCA_MODE_ED -DNRF_802154_CSMA_CA_ENABLED=0 -DNRF_802154_DELAYED_TRX_ENABLED=0 -DNRF_802154_FRAME_TIMESTAMP_ENABLED=0 -DNRF_802154_INTERNAL_RADIO_IRQ_HANDLING=0 -DNRF_802154_IRQ_PRIORITY=1 -DNRF_802154_PENDING_EXTENDED_ADDRESSES=16 -DNRF_802154_PENDING_SHORT_ADDRESSES=16 -DNRF_802154_RX_BUFFERS=8 -DNRF_802154_TX_STARTED_NOTIFY_ENABLED=0 -DNRF_802154_USE_RAW_API=1 -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I../../../kernel/include -I../../../arch/arm/include -I../../../include -I../../../include/drivers -Izephyr/include/generated -I../../../soc/arm/nordic_nrf/nrf52 -I../../../soc/arm/nordic_nrf/include -I../../../lib/libc/minimal/include -I../../../ext/lib/fnmatch/. -I../../../ext/hal/cmsis/Include -I../../../ext/hal/nordic/drivers/nrf_radio_802154/. -I../../../ext/hal/nordic/drivers/nrf_radio_802154/rsch -I../../../ext/hal/nordic/drivers/nrf_radio_802154/rsch/raal -I../../../ext/hal/nordic/nrfx -I../../../ext/hal/nordic/nrfx/drivers/include -I../../../ext/hal/nordic/nrfx/hal -I../../../ext/hal/nordic/nrfx/mdk -I../../../ext/hal/nordic/. -isystem /opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/include -isystem /opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/include-fixed -isystem /home/joaocordeiro/KNoT/openthread/include -isystem /home/joaocordeiro/KNoT/openthread/examples/platforms -Os -nostdinc -g -Wall -Wformat -Wformat-security -Wno-format-zero-length -imacros /home/joaocordeiro/zephyr/samples/hello_world/build/zephyr/include/generated/autoconf.h -ffreestanding -Wno-main -fno-common -mthumb -mcpu=cortex-m4 -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -Wno-pointer-sign -Wno-unused-but-set-variable -fno-reorder-functions -fno-defer-pop -Werror=implicit-int -Wpointer-arith -fmacro-prefix-map=/home/joaocordeiro/zephyr=. -ffunction-sections -fdata-sections -mabi=aapcs -march=armv7e-m -std=c99 -MD -MT zephyr/CMakeFiles/zephyr.dir/drivers/ieee802154/ieee802154_nrf5.c.obj -MF zephyr/CMakeFiles/zephyr.dir/drivers/ieee802154/ieee802154_nrf5.c.obj.d -o zephyr/CMakeFiles/zephyr.dir/drivers/ieee802154/ieee802154_nrf5.c.obj   -c /home/joaocordeiro/zephyr/drivers/ieee802154/ieee802154_nrf5.c
/home/joaocordeiro/zephyr/drivers/ieee802154/ieee802154_nrf5.c: In function 'nrf5_rx_thread':
/home/joaocordeiro/zephyr/drivers/ieee802154/ieee802154_nrf5.c:121:7: error: 'CONFIG_IEEE802154_DRIVER_LOG_LEVEL' undeclared (first use in this function)
   if (CONFIG_IEEE802154_DRIVER_LOG_LEVEL >= LOG_LEVEL_DBG) {
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/joaocordeiro/zephyr/drivers/ieee802154/ieee802154_nrf5.c:121:7: note: each undeclared identifier is reported only once for each function it appears in
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: f78599fbcaebde1dccc34ef0b672fdd894eb9d57

**Additional context**
This happens because nrf5's 802.15.4 driver checks for CONFIG_IEEE802154_DRIVER_LOG_LEVEL even if the Logger is disabled.

jcorde, 2019-05-13:
This happens because of `82d956fdf867fd7eb91472c9b94919657db4d0af`. A possible solution is to check if  CONFIG_IEEE802154_DRIVER_LOG_LEVEL is defined before comparing it, as present at https://github.com/jcorde/zephyr/commit/a8f9c612f2aaebcc8d22455df0e34b9bc512d8b5

rlubos, 2019-05-13:
@jcorde Could you just send the patch you proposed in a PR?

jcorde, 2019-05-13:
@rlubos  Just opened it at https://github.com/zephyrproject-rtos/zephyr/pull/16105