zephyrproject-rtos/zephyr #16122: Detect first block in LWM2M firmware updates.

sunkyl, 2019-05-13:
Problem
Currently, there is no way to detect when to reset the context of a firmware update via LWM2M when the process is interrupted outside of the block received callback (say in the case of a timeout).

Solution
Some indication of a first block for the callback, or exposing more functions within the LWM2M library that allows for writing our own callbacks that can start the transfer. 

Alternatives
Currently, there is the extern function inside the subsystem LWM2M files `lwm2m_firmware_start_transfer()` that I can use. I can create a semi-functional imitation of the sample post-write callback for the package URI that resets a variable to indicate a context reset, but it also has no access to the functions to change the update state. 

Thanks.

mike-scott, 2019-05-15:
@sunkyl Did you want to submit your changes as a PR so we can discuss there while looking at code?

yaomon, 2019-05-24:
Sorry, switching accounts here.
https://github.com/zephyrproject-rtos/zephyr/pull/16369
I think this is the simplest way. I had briefly considered hijacking the instance id parameter in the write callback to pass 1/0 based on whether the current block size is 0 (indicating the first block) but that is very inelegant.
Given access to the block context, I can check whether the current size is <= coap block size to check for the first block and reset the flash context accordingly.

Thanks.

carlescufi, 2020-03-24:
@yaomon could you please submit a new PR?
cc @rlubos

PR #24053 (2020-04-09): net: lwm2m: Publicize firmware block context
This change is to allow access to the firmware block context in order to
give the firmware update callback implementation an indication of when
to reset the flash context. Additionally, it allows for a validity check
between the total expected size downloaded and the actual size
downloaded. A simple implementation can check if the block context's
current downloaded blocks is equal to the expected block size in order
to determine if the flash context needs to be reset. This approach
seemed the simplest, and knowing the firmware block context can have [... cut]
commit a9b92641 (2020-04-09): net: lwm2m: Publicize firmware block context

This change is to allow access to the firmware block context in order to
give the firmware update callback implementation an indication of when
to reset the flash context. Additionally, it allows for a validity check
between the total expected size downloaded and the actual size
downloaded. A simple implementation can check if the block context's
current downloaded blocks is equal to the expected block size in order
to determine if the flash context needs to be reset. This approach
seemed the simplest, and knowing the firmware block context can hav [... cut]
closed 2020-03-24 by carlescufi