zephyrproject-rtos/zephyr #16132: The nRF mesh APP report “Invalid Publish Parameters”

embest-tech, 2019-05-14:
I add a node like this:
static struct bt_mesh_model root_models[] = {
	BT_MESH_MODEL_CFG_SRV(&cfg_srv),
	BT_MESH_MODEL_HEALTH_SRV(&health_srv, &health_pub),
	BT_MESH_MODEL(BT_MESH_MODEL_ID_GEN_ONOFF_CLI, gen_onoff_cli_op,
																		NULL, &onoff[0]),
};

After provistion,sellect Generic Onoff Client.
Do APPKEY BINDING.
Then sellect PUBLICATION ADDRESS, click Apply Publication, the APP throw out a worning box:

           ________________________________________
          | Invalid Publish Parameters                 |
          | The node has reported the publish    |
          | parameters are invalid                        |
          |             OK                                           |
	  |________________________________________|

jhedberg, 2019-05-14:
This isn't necessarily a bug - it's possible nRF Mesh is actually sending parameters which are invalid according to the spec. Could you enable debug logs and report what you see in the console? Which version of Zephyr is this?

embest-tech, 2019-05-14:
@jhedberg 
The parameters are the default that the APP hold.
Ok, I am try to enable the debug messages.
The verstion is v1.10-branch.

And the code is based on samples/bluetooth/mesh. For the user developer, we really need a example code that tell us how to configure the Node after provision.

jhedberg, 2019-05-14:
@embest-tech there have been numerous fixes for Mesh since 1.10, more than I can try to list here. Could you try reproducing the issue with latest master, or if that's not possible with 1.14?

embest-tech, 2019-05-14:
@jhedberg  The reason I choose v1.10 is because the newer version takes too much RAM that it can not be fit for NRF51822.  If I change to the latest version reducing RAM consumption is difficult, any idea on this?

embest-tech, 2019-05-14:
@jhedberg 
I have change to v1.12 , after reduce some RAM consumption. The code can be compile and runs. BUT the same issue come out, I can not add pub for the OnOff client but can add sub for it. It does not make sense. Also, I can add pub and sub for the OnOff server.  I don't know what is the reason for adding pub for a genneric OnOff server.

jhedberg, 2019-05-14:
There have been 84 individual commits to the mesh code since 1.12, so it's still possible that the issue has been fixed since then. Is it the 16 or 32k RAM variant of the 51822 that you have?  You can do basic RAM usage analysis with the "ninja ram_report" command. You could also test your app using the latest Zephyr tree on some other nRF board with more RAM to see if the issue is fixed or not.

jhedberg, 2019-05-14:
I also wanted to mention some typical ways of reducing the RAM usage:

 - Reduce stack sizes to minimum of what's needed (there are Kconfig options to help with current usage analysis)
 - Reduce buffer counts and sizes
 - Disable unneeded features in Kconfig

embest-tech, 2019-05-15:
@jhedberg 
Thanks for you warm help. The device we use is 16KB version. We have started to purchase Nordic nRF52840-PDK board,which are mentioned in zephyr document, to help to analyse this issue. But we tend to use the low power M0 core for the real product.  You are the expert for the mesh stack so I'd like to ask if there is potential for reducing the RAM usage to a proper scale, such as from 90% to 70%?

embest-tech, 2019-05-15:
@jhedberg   
The problerm has been fixed. It's my mistake, gen_onoff_cli_op need to be add to the model which want to publish.

BT_MESH_MODEL_PUB_DEFINE(**gen_onoff_pub_cli**, NULL, 2 + 2);

static struct bt_mesh_model root_models[] = {
	BT_MESH_MODEL_CFG_SRV(&cfg_srv),
	//BT_MESH_MODEL_HEALTH_SRV(&health_srv, &health_pub),
	BT_MESH_MODEL(BT_MESH_MODEL_ID_GEN_ONOFF_CLI, **gen_onoff_cli_op**,
																		&gen_onoff_pub_cli, &onoff[0]),
};

jhedberg, 2019-05-15:
> The problerm has been fixed. It's my mistake, gen_onoff_cli_op need to be add to the model which want to publish.

@embest-tech good to hear, I'll close this issue then.

Regarding mesh footprint reduction, it really depends on what your current configuration is and what features you have enabled. I've been through the exercise of fitting a 16k nRF51 target in the past with the BBC micro:bit board. The sample app used for that was `samples/bluetooth/mesh_demo`. You could e.g. try looking at its `prj_bbc_microbit.conf` file to see what kind of options were tuned. It's worth mentioning however that it disables all provisioning support completely, which saves up to 1k of RAM or so, which is probably not something you'll be able to do.

Regarding your comment on power consumption, I've understood that nRF52 can actually be more power efficient since it's able to reach idle state faster (due to a faster CPU), however it'd be better get some official comment on this from a Nordic representative.

closed 2019-05-15 by jhedberg