zephyrproject-rtos/zephyr #16138: is this right for clock announcing in every CORE?

thomascp, 2019-05-14:
Hi,

Actually, I have some questions on SMP, I am not sure it is a defect.
Take x86_64 SMP as an example, hpet_isr will execute on every CORE and it will call z_clock_announce, is this right for clock announcing in every CORE?
You know, in Linux, only one core is responsible for do_timer which will increase the jiffies, so...

And, imagine there are two cores, core1 is running a thread with A priority, core2 is running a thread with B priority, an interrupt hit core1, and let us say A priority is higher than B, the interrupt give a semaphore to a task whose priority is lower than A and higher than B, what will happen? As I see in the codes, it seems core1 which is running a thread with A will not give up the scheduler, but core2 doesn't know about this interrupt, is this correct? Should we tell the core2?

BR,
Peng

andyross, 2019-05-14:
Yes, HPET interrupt is global and delivers to every core by default.   The ISR takes a spinlock, so only one CPU will see a time delta and call z_clock_announce() with a non-zero argument.

I experimented briefly with setting external interrupts to "deliver to lowest priority CPU" in the ioapic, but that has a side effect of significantly complicating timeslicing, which has to share the same timer interrupt.  It's definitely an area that could be better optimized

andyross, 2019-05-14:
Oh, sorry, and there's a second part to your question: no, there is no scheduler IPI made on changes to the ready queue.  You'll get a synchronous wakeup of a thread on your CPU only, the other CPU will need to wait for another interrupt.  The alternative would be to interrupt all CPUs in the system any time any thread woke up, which is simply too expensive.

Note that there is (on x86_64) a scheduler IPI mechanim defined.  In theory we could expose this as an API for applications that want to exert that kind of control over the scheduler in an SMP environment.   Right now, it's used only by k_thread_abort() (which for obvious reasons has to be synchronous and interrupt a running thread).

thomascp, 2019-05-14:
Hi andyross,

Thank you for your quick reply!
1. The ISR takes a spinlock, so only one CPU will see a time delta and call z_clock_announce() with a non-zero argument.
You mean it works with the configuration of CONFIG_TICKLESS_KERNEL, but without it, every core which enters the ISR will announce 1 tick in the ISR routine.
Actually, I am not familiar with IA arch, so maybe for the hept,  CPU can promise that only one core get into the timer ISR? Then it is OK.
2. For the second question, it is not a good example for priority based OS,  I think there is a simple way to solve this. Every time when an interrupt happens,  check the others core's current to see whether it need to be rescheduled and then send them a well defined IPI.

BR,
Peng

andyross, 2019-05-17:
Ah, indeed. #1 is a bug.  Looks to me like the clock is going to run Ncpu times as fast as it should if you enable SMP but not tickless (which is sort of pathological -- ticked kernels at this point are for resource constrained systems that don't want to pay for the extra complexity in the ISR).

As far as #2 goes: that's doable if you have two CPUs, but it doesn't scale well -- every scheduler changes requires a loop over all CPUs.   IMHO if we have an app with requirements like this (that the extra CPUs be perfectly synchronous with respect to preemption events) we should just provide the IPI hook as an API to the app so it can make the decision.

closed 2019-06-13 by thomascp