zephyrproject-rtos/zephyr #16141: posix: CONFIG_POSIX_API and CONFIG_SOCKETS_POSIX_NAMES don't make sense to use together, and conflict when doing so

pfalcon, 2019-05-14:
As discussed in https://github.com/zephyrproject-rtos/zephyr/pull/16009 .

The meaning of CONFIG_SOCKETS_POSIX_NAMES is "expose Zephyr's namespaced socket API under standard POSIX names".

The meaning of CONFIG_POSIX_API is "expose all POSIX APIs which are available in Zephyr".

From this description, it's clear that these 2 options shouldn't be used together, as CONFIG_POSIX_API has wider scope which subsumes and supersedes meaning of CONFIG_SOCKETS_POSIX_NAMES. 

However, implementation-wise, these 2 options are implemented differently, so if someone by mistake enabled them, that can lead to compile-time errors (whose reason may be not immediately clear.

Consequently, we should:

* Either make these 2 options conflict, with explicit error message describing the above, or
* Make CONFIG_POSIX_API silently subsume CONFIG_SOCKETS_POSIX_NAMES.

carlescufi, 2019-05-14:
Relabeling as enhancement since this is about the meaning of Kconfig options.

pfalcon, 2019-05-14:
> Relabeling as enhancement since this is about the meaning of Kconfig options.

Interesting, then why @nashif blocks a legitimate pull request https://github.com/zephyrproject-rtos/zephyr/pull/16009 , based on this unrelated enhancement?

PR #16557 (2019-06-05): net: sockets: Make NET_SOCKETS_POSIX_NAMES depend on !POSIX_API
These options are mutually exclusive, or more specifically,
CONFIG_POSIX_API has wider scope and supersedes
CONFIG_NET_SOCKETS_POSIX_NAMES. Implementation-wise, the two
options should not be defined at the same time, as that may
lead to declaration conflicts.

Fixes: #16141
commit 0fb8a917 (2019-06-05): net: sockets: Make NET_SOCKETS_POSIX_NAMES depend on !POSIX_API

These options are mutually exclusive, or more specifically,
CONFIG_POSIX_API has wider scope and supersedes
CONFIG_NET_SOCKETS_POSIX_NAMES. Implementation-wise, the two
options should not be defined at the same time, as that may
lead to declaration conflicts.

Fixes: #16141
closed 2019-06-05 by nashif