zephyrproject-rtos/zephyr #16143: posix: clock_settime calculates the base time incorrectly

alexanderendian, 2019-05-14:
If you call `clock_settime` twice with the same non-zero value appears to reset the clock back to 0. I've that disovered that it's caused by `clock_settime` performing its internal calculation incorrectly when updating the clock base. I have submitted a pull request which contains some further explanation and a fix #16040.

**To Reproduce**

1. Build and run the following program in QEMU (although it should affect any target). _Disclaimer: I don't have a development environment nearby to test it, so it may contain a typo._
```C
#include <zephyr.h>
#include <posix/time.h>
#include <misc/printk.h>

void main(void)
{
	struct timespec target;
	struct timespec actual;

	target.tv_sec = 1000000;
	target.tv_nsec = 0;

	printk("clock_settime to %d\n", target->tv_sec);
	clock_settime(CLOCK_REALTIME, &target);
	clock_gettime(CLOCK_REALTIME, &actual);
	printk("clock_gettime returned %d\n", actual->tv_sec);

	printk("clock_settime to %d\n", target->tv_sec);
	clock_settime(CLOCK_REALTIME, &target);
	clock_gettime(CLOCK_REALTIME, &actual);
	printk("clock_gettime returned %d\n", actual->tv_sec);
}
```
2. Observe that the last printed line shows a time close to 0 when something close to the target of 1000000 is expected.

galak, 2019-05-14:
@alexanderendian thanks for submitting this issue.

jbrzozoski, 2019-05-17:
I *just* bumped into this, too.  Fix is just to use CLOCK_MONOTONIC when getting the base reference inside clock_settime.

PR #16040 (2019-05-21): posix: Fix calculation of clock base in clock_settime
Previous version calculated rt_clock_base incorrectly by subtracting
clock_gettime from the specified time. Effectively the following
formula was used.

    rt_clock_base := new_time - clock_gettime()

This is clearly incorrect when we consider what should happen if we
call clock_settime with the result of clock_gettime. It ought to be
approximately a no-op, but instead we end up zeroing the clock.

    rt_clock_base := clock_gettime() - clock_gettime() = 0

This patch fixes clock_settime by instead using k_uptime_get to
calculate rt_clock_ [... cut]
PR #16283 (2019-05-21): [Backport v1.14-branch] posix: Fix calculation of clock base in clock_settime
Backport #16040.

Fixes: #16143
closed 2019-05-21 by nashif