zephyrproject-rtos/zephyr #16150: Make more LWM2M parameters configurable at runtime

lodup29, 2019-05-14:
**Description**
Some LWM2M parameters which one would expect to be able to configure at runtime can only be configured at compile time:
- CONFIG_LWM2M_FIRMWARE_UPDATE_PULL_COAP_PROXY_ADDR
- CONFIG_LWM2M_PEER_PORT

To provide an API to override these, possibly through lwm2m_ctx, would seem like a convenient fit.

carlescufi, 2019-05-15:
@lodup29 this sounds like a very good first issue for a Pull Request. Would you like to give it a try?

lodup29, 2019-05-15:
@carlescufi sure. I'd just like to hear from the lead developer (Mike Scott) fist to make sure that lwm2m_ctx is the proper place for LWM2M specific parameters.

mike-scott, 2019-05-15:
@lodup29 I agree that `CONFIG_LWM2M_FIRMWARE_UPDATE_PULL_COAP_PROXY_ADDR` is probably something that we should save int the context and be able to change if needed.  However, `CONFIG_LWM2M_PEER_PORT` is just the default port used when no port is explicitly set in the connection URL:
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/lib/lwm2m/lwm2m_engine.c#L4080

I don't have a strong objection to it, but I'm not sure it's necessary.

mike-scott, 2019-05-15:
@lodup29 Feel free to setup a PR with this change and we can discuss there.

lodup29, 2019-05-18:
@mike-scott Indeed I missed that. The UDP port can be set through the URL by appending ":UDP_PORT". I see no reason to dissociate the two.
However the client example is a little misleading as it checks for the ':' character to detect if it is an IPV6 address and if so it will encapsulate the whole thing (including the port) within [...]. How about changing the client demo to build the server URL as:
```C
	snprintk(server_url, server_url_len, "coap%s//%s%s%s:%d",
		 IS_ENABLED(CONFIG_LWM2M_DTLS_SUPPORT) ? "s:" : ":",
		 strchr(SERVER_ADDR, ':') ? "[" : "", SERVER_ADDR,
		 strchr(SERVER_ADDR, ':') ? "]" : "",
		 CONFIG_LWM2M_PEER_PORT);
```
I feel it would be more representative of the usage.

lodup29, 2019-05-26:
@mike-scott Unfortunately the lwm2m_ctx used in lwm2m_obj_firmware_pull.c in recreated from scratch and cannot really be used to carry parameters there. I simply added a new function (lwm2m_firmware_set_proxy) to set the proxy at runtime. I submitted the pull request [16396](https://github.com/zephyrproject-rtos/zephyr/pull/16396) accordingly.

rlubos, 2021-02-15:
Closing due to lack of activity in the issue/PR for almost 2 years.

closed 2021-02-15 by rlubos