zephyrproject-rtos/zephyr #16154: Fix various issues with handling of floating value within the LWM2M subsystem

lodup29, 2019-05-14:
**Describe the bug**
- Regardless of its representation (_TLV_, _JSON_), it's not possible to pass a value > -1.0 and < 0.0 (ex: -0.5).  There's no _sign_ field in the [_float32_value_](https://github.com/zephyrproject-rtos/zephyr/blob/5d8e367efe23915a879620bcec61b0d7c1c0c1e0/include/net/lwm2m.h#L345) and [_float64_value_](https://github.com/zephyrproject-rtos/zephyr/blob/5d8e367efe23915a879620bcec61b0d7c1c0c1e0/include/net/lwm2m.h#L358) so the only way to represent a value such as -0.5 is :
```c
struct float32_value lFloatValue; /* val1  and val2 are integers (s32_t) */
lFloatValue.val1 = 0;
lFloatValue.val2 = -500000;
```
However [only the sign of _val1_ is considered for _TLV_](https://github.com/zephyrproject-rtos/zephyr/blob/5d8e367efe23915a879620bcec61b0d7c1c0c1e0/subsys/net/lib/lwm2m/lwm2m_util.c#L79) while the _JSON_ representation simply formats the value as [_("%d.%d",lFloatValue.val1, lFloatValue.val2)_](https://github.com/zephyrproject-rtos/zephyr/blob/5d8e367efe23915a879620bcec61b0d7c1c0c1e0/subsys/net/lib/lwm2m/lwm2m_rw_json.c#L500) ("0.-500000") thus resulting in an exception when attempting to parse the value on the other side.
- Using the _JSON_ representation it is not possible to represent a value with leading zeroes for _val2_ (ex: 0.05). Indeed as such value should be reprensented as:
```c
struct float32_value lFloatValue; /* val1  and val2 are integers (s32_t) */
lFloatValue.val1 = 0;
lFloatValue.val2 = 50000; /* 0.05*1000000 */
```
However as previously stated the value will simply end up being formated as [_("%d.%d",lFloatValue.val1, lFloatValue.val2)_](https://github.com/zephyrproject-rtos/zephyr/blob/5d8e367efe23915a879620bcec61b0d7c1c0c1e0/subsys/net/lib/lwm2m/lwm2m_rw_json.c#L500) ("0.500000" instead of "0.050000")
- At least some mantissas are one bit too short using the TLV reprensentation. Trying the represent 0.05 (0x3D4CCCCD using IEEE 754 single precision) ends up as 0.049999997 (0x3D4CCCCC) on the other side

**To Reproduce**

- Use the _Leshan_ lwm2m server demo as the server backend
- Select _Single-value_ in the top right drop-down menu as _JSON_ or _TLV_ based on the test case
- Use the zephyr lwm2m client example and change the value of the temperature object as described by the following cases:

| Representation  | val1 | val2 | Expected value | Value retrieved by _Leshan_ |
| ------------- | ------------- | ------------- |------------- |------------- |
| _TLV_  | 0  | -500000 | -0.5 | 0.5 |
| _JSON_ | 0  | -500000 | -0.5 | Exception |
| _JSON_ | 0  | 50000 | 0.05 | 0.5 |
| _TLV_  | 0  | 50000 | 0.05 | 0.049999997 |

ioannisg, 2019-08-17:
@mike-scott, same question as before; wonder if this is still applicable, and, if so, if it could be addressed for 2.0

mike-scott, 2019-08-17:
Thank you for the reminder.  I will address.

ioannisg, 2019-08-27:
> Thank you for the reminder. I will address.

Just asking if this has this already been addressed, @mike-scott - few LWM2M Pr were merged lately..

closed 2019-08-28 by ioannisg