zephyrproject-rtos/zephyr #16158: LWM2M : last_block is not set through firmware write callback

lodup29, 2019-05-14:
At least using a coap to http proxy (I never manager to trigger the callback otherwise) the _last_block_ parameter provided though the callback function (implemented through _firmware_block_received_cb_ in the LWM2M client example) is never set to true although the firmware is fully downloaded. 

**To Reproduce**
- Build and run a firmware distribution point using _docker-compose_:
```bash
version: '3'

services:
  cf-proxy-coap-http:
    image: hub.foundries.io/cf-proxy-coap-http:latest
    container_name: coap-proxy
    restart: unless-stopped
    ports:
      - "5682:5682/udp"
  nginx:
    image: opensourcefoundries/nginx:latest
    container_name: firmware-server
    restart: unless-stopped
    ports:
      - "8080:80"
    volumes:
      - /home/admin-local/data:/usr/share/nginx/html
```
- Add support for firmware pull using COAP proxy to the LWM2M client example:
```C
CONFIG_LWM2M_FIRMWARE_UPDATE_PULL_SUPPORT=y
CONFIG_LWM2M_FIRMWARE_UPDATE_OBJ_SUPPORT=y
CONFIG_LWM2M_FIRMWARE_UPDATE_PULL_COAP_PROXY_SUPPORT=y
CONFIG_LWM2M_FIRMWARE_UPDATE_PULL_COAP_PROXY_ADDR="coap://FIRMWARE_IP_ADDRESS:5682"
CONFIG_LWM2M_COAP_BLOCK_SIZE=1024
```
and monitor _last_block_.

- Run the _Leshan demo server_

```bash
wget https://hudson.eclipse.org/leshan/job/leshan/lastSuccessfulBuild/artifact/leshan-server-demo.jar
sudo java -jar ./leshan-server-demo.jar -wp 80
```
- Place a file named _update.bin_ in the directory mounted on _/usr/share/nginx/html_ (see docker-compose configuration above) and write _http://FIRMWARE_IP_ADDRESS:8080/update.bin_ to _/5/0/1_ (_Firmware Update/Instance 0/Package URI_)

The download should go fine but _last_block_ is never set to to true.

mike-scott, 2019-05-15:
@lodup29 Thank you for bringing this up.  It might be that over time the logic to the last_block handling has been broken (from the callback perspective).

We calculate last_block here:
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/lib/lwm2m/lwm2m_obj_firmware_pull.c#L295

And then check for a write callback here:
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/lib/lwm2m/lwm2m_obj_firmware_pull.c#L295

With the actual callback happening here:
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/lib/lwm2m/lwm2m_obj_firmware_pull.c#L337

As I get some time, I'll run a few tests and see what needs to happen there.

ioannisg, 2019-08-17:
@mike-scott is this bug still present? If so, will you address this for 2.0.0?

mike-scott, 2019-08-17:
Yes, I will test and address for 2.0 if there's a bug.

ioannisg, 2019-08-27:
@mike-scott ping :)

closed 2019-08-29 by ioannisg