zephyrproject-rtos/zephyr #16177: STM32: Could not compile with CONFIG_PINMUX=n

erwango, 2019-05-15:
**Describe the bug**
STM32 targets won't compile with CONFIG_PINMUX=n.

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/basic/minimal
2. west build -b nucleo_f334r8 -- -DOVERLAY_CONFIG="common.conf no-mt.conf no-timers.conf arm.conf"
3. See error

**Expected behavior**
Compilation ok

**Impact**
Minimal build ko

**Screenshots or console output**
```
/local/mcu/zephyr-sdk-0.10.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: Undefined initialization levels used.
/local/mcu/zephyr-sdk-0.10.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: zephyr/boards/arm/nucleo_f334r8/libboards__arm__nucleo_f334r8.a(pinmux.c.obj): in function `pinmux_stm32_init':
/local/mcu/zephyr/zephyr-project/boards/arm/nucleo_f334r8/pinmux.c:48: undefined reference to `stm32_setup_pins'
collect2: error: ld returned 1 exit status
```

**Environment (please complete the following information):**
 - OS: Ubuntu16.04
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Could be solved with following patch in each stm32 board CMakeLists.txt
```
+ if(CONFIG_PINMUX)
   zephyr_library()
   zephyr_library_sources(pinmux.c)
   zephyr_library_include_directories(${ZEPHYR_BASE}/drivers)
+ endif()
```

carlescufi, 2019-05-15:
I actually hit this yesterday! I was writing a minimal sample (`samples/basic/minimal`) and that did not work with ST boards because it disables PINMUX

erwango, 2019-05-15:
> I actually hit this yesterday! I was writing a minimal sample (samples/basic/minimal) and that did not work with ST boards because it disables PINMUX

:-), I hit this when testing your minimal sample this morning

PR #16618 (2019-06-10): Fix16177 STM32: Could not compile with CONFIG_PINMUX=n
Fix #16177 STM32: Could not compile with CONFIG_PINMUX=n
By the way, fix also minor misalignment within STM32 boards.
commit e1de4cf6 (2019-06-10): boards: Set pinmux.c compilation under switch CONFIG_PINMUX

Fix compilation issue for STM32 boards with CONFIG_PINMUX=n
Fixes #16177
commit f1bf04ad (2020-01-07): boards: stm32f030_demo: Set pinmux.c compilation under switch CONFIG_PINMUX

Based on this commit

  | commit e1de4cf6b549318d7c09d37a84e17c17e2d628ab
  | Author: Alexandre Bourdiol <alexandre.bourdiol@st.com>
  | Date:   Thu Jun 6 15:47:23 2019 +0200
  |
  |     boards: Set pinmux.c compilation under switch CONFIG_PINMUX
  |
  |     Fix compilation issue for STM32 boards with CONFIG_PINMUX=n
  |     Fixes #16177
closed 2019-06-10 by carlescufi