zephyrproject-rtos/zephyr #16195: k_uptime_delta(): Defective by design

pizi-nordic, 2019-05-16:
The k_uptime_delta() API is widely used in Zephyr tests. However due to flaw in the design, the values returned by k_uptime_delta() depends on the uptime of the system if (1000 / CONFIG_SYS_CLOCK_TICKS_PER_SEC) is not an integer. As result, multiple tests are failing (example: https://github.com/zephyrproject-rtos/zephyr/issues/15983).

Cause:
The k_uptime_delta() calculates delta time using two timestamps (expressed in milliseconds) obtained form the system uptime (measured in system ticks). Unfortunately, the uptime conversion from ticks to milliseconds is inaccurate when (1000 / CONFIG_SYS_CLOCK_TICKS_PER_SEC) is not an integer. The rounding, which is applied in such case, leads to errors in delta time calculation.

Example (for CONFIG_SYS_CLOCK_TICKS_PER_SEC = 128):

Tick | Uptime [ms] | k_uptime_get() | Delta time from previous tick returned by k_uptime_delta()
-- | -- | -- | --
0 | 0,000 | 0 | 
1 | 7,813 | 7 | 7
2 | 15,625 | 15 | 8
3 | 23,438 | 23 | 8
4 | 31,250 | 31 | 8
5 | 39,063 | 39 | 8
6 | 46,875 | 46 | 7
7 | 54,688 | 54 | 8
8 | 62,500 | 62 | 8
9 | 70,313 | 70 | 8
10 | 78,125 | 78 | 8

Please note, that ticks 1 and 6 are reported as shorter than others, which is not true. 

Possible solution:
The k_uptime_delta() API in current form cannot be fixed. The reference time has to be stored in ticks (or other unit used by kernel) and all calculations should be performed using this unit. Only the result of calculations should be converted to ms.

pizi-nordic, 2019-05-16:
FYI: @andyross, @nashif, @carlescufi, @pabigot

pabigot, 2020-09-08:
Per the original description `k_uptime_delta()` can't be fixed.  A potential solution is to add `k_cycle_delta()` and recommend it as an alternative.  Parallel to that adding `uint64_t k_cycle_get()` as a companion to `uint32_t k_cycle_get_32()` would eliminate a functionality gap.  However there probably isn't support for 64-bit cycle count on all platforms.

So `k_tick_delta()` along with `static inline uint64_t k_tick_get() { return z_tick_get();}` might be required to complete the functionality for long durations.

closed 2020-11-22 by github-actions[bot]