zephyrproject-rtos/zephyr #16196: display_mcux_elcdif driver full support frame buffer features

hakehuang, 2019-05-16:
when I enable the samples/display cases for RT1050 boards ( patch can be found https://github.com/hakehuang/zephyr/commit/986880ba138e6c8f9e88634fdd239b25d7aa19ac) 
there are many frame buffer functions that are required by cfb/cfb_customer_font/cfb_shell samples

propose to add framebuffer function support for elcd driver.
below framebuffer APIs are better to have
{code}
display_set_pixel_format:
    need add all available format. in CFB cases the PIXEL_FORMAT_MONO10, and  
   SCREEN_INFO_MONO_VTILED shall be support
display_blanking_off:
mcux_elcdif_get_framebuffer:
mcux_elcdif_get_capabilities:
     shall add SCREEN_INFO_MONO_VTILED support
{code}

hakehuang, 2019-05-16:
@MaureenHelm

jfischer-phytec-iot, 2019-05-16:
cfb is designed for the monochrome displays, there are other things i would like to improve, support for rgb pannels has low priority.

hakehuang, 2019-05-17:
in this case, do we have any plan to add a general framebuffer support?

carlescufi, 2020-03-24:
@vanwinkeljan @jfischer-phytec-iot is this still relevant?

vanwinkeljan, 2020-03-24:
I don't think so cfb is mainly targeting monochrome displays and mcux elcdif can use lvgl.

closed 2020-03-25 by hakehuang