zephyrproject-rtos/zephyr #16202: Improve help for west build target

galak, 2019-05-16:
The current help for `west build -h` gets (for `target`):

```
  -t TARGET, --target TARGET
                        Build system target to run
```

It wasn't at all obvious at first reading what `target` meant here.  Since this is Zephyr specific, I think it would be useful to have an example for how to invoke the 'run' target for qemu. (unfortunately double use of run)

mbolivar, 2019-05-16:
> It wasn't at all obvious at first reading what target meant here. Since this is Zephyr specific, I think it would be useful to have an example for how to invoke the 'run' target for qemu. (unfortunately double use of run)

The use of `west build -t run` for QEMU is documented in the getting started guide:

https://docs.zephyrproject.org/latest/getting_started/index.html#run-the-application-in-qemu

```
Using west:

west build -b qemu_x86 samples/hello_world
west build -t run
```

Several other usage examples are documented in the page for west build:

https://docs.zephyrproject.org/latest/guides/west/build-flash-debug.html#basics

```
To specify the build system target to run, use --target (or -t).

For example, on host platforms with QEMU, you can use the run target to build and run the Hello World sample for the emulated qemu_x86 board in one command:

west build -b qemu_x86 -t run samples/hello_world
As another example, to use -t to list all build system targets:

west build -t help
As a final example, to use -t to run the pristine target, which deletes all the files in the build directory:

west build -t pristine
```

What exactly is missing here? The `west build -h` output is meant to be a short overview, with full details in the online help.

mbolivar, 2019-05-16:
See also https://github.com/zephyrproject-rtos/zephyr/issues/5501 for a related open issue related to the QEMU devices.

I didn't do it because I already faced enough backlash about "forcing" people to use west, and those targets have been special-cased and done their own thing since the Kbuild days.

I think most of that pushback has been resolved, though, and I've heard from 4 people in the past month or so that they'd like to see more west integration with the runners, so I guess the "yeas" are starting to outvote the "nays" on this.

galak, 2019-05-16:
> What exactly is missing here? The `west build -h` output is meant to be a short overview, with full details in the online help.

I guess I'm hoping that the help we get from the tool is a bit more useful than having to go to online pages.

mbolivar, 2019-05-20:
> I guess I'm hoping that the help we get from the tool is a bit more useful than having to go to online pages.

The problem is that if we try to do that, the `-h` output will be way too long for a command line tool, in my opinion.

There are a lot of tools that  make this tradeoff in my experience (short `foo -h` and details in `man foo` or the online help).

commit 150eb657 (2019-12-19): scripts: west build: slight --help tweak

We have an open request to make the help for the -t option a little
easier. Try to do that without adding too much length to the short
help.

Fixes: #16202
closed 2019-12-19 by carlescufi