zephyrproject-rtos/zephyr #16204: Build STM32 : generate hex file fail

Benichou34, 2019-05-16:
**Describe the bug**
Build displays the following errors messages:
```
C:\Program Files (x86)\GNU Tools ARM Embedded\8 2018-q4-major\bin\arm-none-eabi-objcopy.exe: zephyr.hex 64-bit address 0x4b4fa308000000 out of range for Intel Hex file
C:\Program Files (x86)\GNU Tools ARM Embedded\8 2018-q4-major\bin\arm-none-eabi-objcopy.exe:zephyr.hex: bad value
```
**To Reproduce**
Steps to reproduce the behavior: Build "Hello world" sample with nucleo_F413zh board.

**Expected behavior**
Generate hex file.

**Environment:**
 - OS: Windows 10 x64
 - Toolchain: GNU ARM toolchain 8-2018-q4-major
- Ninja 1.9.0.20190208
- CMake 3.14.4

SebastianBoe, 2019-05-16:
Duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/12257

closed 2019-05-16 by SebastianBoe

Benichou34, 2019-05-16:
Hello @SebastianBoe,
Please note that since PR #16022, generation of hex file for STM32 is activated by default.
So, this issue appears all the time, for each build.

SebastianBoe, 2019-05-20:
Noted.