zephyrproject-rtos/zephyr #16205: Add support to west to flash w/o a build, but given a binary

galak, 2019-05-16:
Support something like:

`west flash -b <BOARD> [-r RUNNER] <FILENAME>`

Which should allow flashing the board with <FILENAME> w/o doing any type of build.

This is kinda related to:

https://github.com/zephyrproject-rtos/west/issues/124

mbolivar, 2019-05-16:
Something like this could definitely be done on a per-runner basis. I'm not sure offhand if it can be done in general; we rely on the build directory for various bits of context necessary to flash, but we should be able to pull that from the board directory without doing a build in some cases.

mbolivar-nordic, 2022-05-25:
I'm going to close this as WONTFIX. As time has passed, it's become more and more clear that the west runners need the extra context in a build directory to do their work. I think flashing arbitrary binary files without a build directory (as opposed to overriding some artifacts with e.g. `west flash -r nrfjprog --hex-file somethingelse.hex`) is out of scope for this package.

closed 2022-05-25 by mbolivar-nordic