zephyrproject-rtos/zephyr #16211: NVS: sector erase at startup

kapi-no, 2019-05-16:
I am trying to understand the NVS code and evaluate the stability of this module. In this process, I stumbled across logic at the end of nvs_startup() function, which seems suspicious to me. Here is the relevant code excerpt:

	/* if the sector after the write sector is not empty gc was interrupted
	 * we need to restart gc, first erase the sector before restarting gc
	 * otherwise the data may not fit into the sector.
	 */
	addr = fs->ate_wra & ADDR_SECT_MASK;
	nvs_sector_advance(fs, &addr);
	rc = nvs_flash_cmp_const(fs, addr, 0xff, fs->sector_size);
	if (rc < 0) {
		goto end;
	}
	if (rc) {
		/* the sector after fs->ate_wrt is not empty */
		rc = nvs_flash_erase_sector(fs, fs->ate_wra);
		if (rc) {
			goto end;
		}
		fs->ate_wra &= ADDR_SECT_MASK;
		fs->ate_wra += (fs->sector_size - 2 * ate_size);
		fs->data_wra = (fs->ate_wra & ADDR_SECT_MASK);
		rc = nvs_gc(fs);
		if (rc) {
			goto end;
		}
	}

At this point, ```fs->ate_wra``` is pointing to the next ATE write address of the first open sector. Assuming that this sector has some valid ATEs, consider this line from the code excerpt:

```
rc = nvs_flash_erase_sector(fs, fs->ate_wra);
```

Wouldn't this erase the valid ATEs? Or am I missing something?

kapi-no, 2019-05-16:
CC: @Laczen

nvlsianpu, 2019-05-16:
cc @lemrey

Laczen, 2019-05-16:
@kapi-no, the idea is the following: when it finds data in the next sector it was doing a gc from the next sector to the present sector. So to keep it simple we just erase the present sector and restart the gc operation. The valid ate's that were in the present sector will be written again by the gc operation.

How is the stability of the code evaluation ?

nvlsianpu, 2019-05-16:
So there are sequence  of sectors:
(1st) write sector
(2nd) dirty by not finished GC destination sector
(3rd) closed sector which was the source of not finished GC

As I understand in case such situation was detected the 2nd sector should be erased, and then GC should be called again. Looks like the 1st sector is erased.

How is the stability of the code evaluation ?
We have been observing data lost - now investigate for a bug source.

Laczen, 2019-05-16:
> So there are sequence of sectors:
> (1st) write sector
> (2nd) dirty by not finished GC destination sector
> (3rd) closed sector which was the source of not finished GC.
>
> As I understand in case such [...]

There are only 2 sectors: write sector (also the dirty sector) and the next (closed) sector. The write sector only contains data that is also in the next sector

> How is the stability of the code evaluation ?
> We have been observing data lost - now investigate for a bug source.

That is not good, do you mean data loss or data not written ? In the case of gc interruption you can expect the data that was meant to be written to be lost as this will only be done after gc has been completed.

kapi-no, 2019-05-17:
> There are only 2 sectors: write sector (also the dirty sector) and the next (closed) sector. The write sector only contains data that is also in the next sector

Where is it ensured that the next (closed) sector has the same data as the previous one (the dirty one)?

And thanks for some explanation so far.

Laczen, 2019-05-17:
> > There are only 2 sectors: write sector (also the dirty sector) and the next (closed) sector. The write sector only contains data that is also in the next sector
> 
> Where is it ensured that the next (closed) sector has the same data as the previous one (the dirty one)?
>

This is done in the nvs_write() routine. The way this works is: is there sufficient space to write the new data, if not first do gc. This ensures that the gc operation is starting in a fresh empty sector: so all data in the gc operation is coming from the closed (last sector). Ones the gc operation is finished the write of new data will be performed (unless there is no more space available).

> And thanks for some explanation so far.

Your welcome

kapi-no, 2019-05-17:
Here is another question from me (also related to the startup function):

	/* possible data write after last ate write, update data_wra */
	while (1) {
		empty_len = fs->ate_wra - fs->data_wra;
		if (!empty_len) {
			break;
		}
		rc = nvs_flash_cmp_const(fs, fs->data_wra, 0xff, empty_len);
		if (rc < 0) {
			goto end;
		}
		if (!rc) {
			break;
		}
		fs->data_wra += fs->write_block_size;
	}

Why is this piece of logic necessary? It seems that this fragment loops over the data which are not associated with any ATE. Why do we want to preserve it by overriding ```data_wra```?

Laczen, 2019-05-17:
@kapi-no, the nvs_write function will first write the data and afterwards the ate. If the ate for whatever reason does not get written, the data will already be there. This is data without an ate. This region cannot be used for writing data because it already has data and you cannot write the flash twice. So at startup it is checked if there are regions like this and the data_wra is updated just after this region.

Laczen, 2019-05-18:
@kapi-no, @nvlsianpu It was mentioned that data loss was discovered. This seems like a big bug to me, could you please elaborate this ?

nvlsianpu, 2019-05-20:
This take so much time as we ware trying to find the bug root.

We used 2 sector for NVS.

After first GC operation the 1st sector become scratch and the 2nd sector become write sector. And when, after that NVS was initialize (via reboot) again - it recognized 1st sector as write sector and 2nd as undone GC destination sector, so it cleared 2nd sector and  re-run GC, which causes data loss.

Laczen, 2019-05-20:
I think I have found what is causing this: in the startup there is the code:

```
for (u16_t i = 0; i < fs->sector_count; i++) {
	addr = (i << ADDR_SECT_SHIFT) + fs->sector_size - ate_size;
	rc = nvs_flash_cmp_const(fs, addr, 0xff,
					  sizeof(struct nvs_ate));
	if (rc) {
		/* closed sector */
		nvs_sector_advance(fs, &addr);
		rc = nvs_flash_cmp_const(fs, addr, 0xff,
					  sizeof(struct nvs_ate));
		if (!rc) {
			/* open sector */
			break;
		}
	}
	/* none of the sectors where closed, set the address to
	 * the first sector
	 */
	nvs_sector_advance(fs, &addr);
}
```

If none of the sectors were closed you can normally set the address to the first sector, this is not valid for a system with only two sectors. In the case of two sectors you will always have the case that none of the sectors are closed.

nvlsianpu, 2019-05-20:
^^ Yes that is my judgment as well.

nvlsianpu, 2019-05-20:
I will send patch a fix this.

kapi-no, 2019-05-20:
I have extended the NVS test suite to cover this scenario in this PR: https://github.com/zephyrproject-rtos/zephyr/pull/16255. The test that I have added should fail til this bug is fixed.

Laczen, 2019-05-20:
A simple patch could be:
```
	for (u16_t i = 0; i < fs->sector_count; i++) {
		addr = (i << ADDR_SECT_SHIFT) + fs->sector_size - ate_size;
		rc = nvs_flash_cmp_const(fs, addr, 0xff,
					  sizeof(struct nvs_ate));
		if (rc) {
			/* closed sector */
			nvs_sector_advance(fs, &addr);
			rc = nvs_flash_cmp_const(fs, addr, 0xff,
						  sizeof(struct nvs_ate));
			if (!rc) {
				/* open sector */
				break;
			}
		}
		/* none of the sectors where closed, in most cases we can set
		 * the address to the first sector, except when there are only
		 * two sectors. Then we can only set it to the next sector if
		 * the last sector contains no ate's. So we check this first 
		 */
		rc = nvs_flash_cmp_const(fs, addr - ate_size, 0xff,
					  sizeof(struct nvs_ate));
		if (rc) {
			/* we found data */
			break;
		}
		nvs_sector_advance(fs, &addr);
	}
```

kapi-no, 2019-05-20:
Maybe, we should move it outside this loop so this doesn't affect performance when more than 2 sectors are defined? We can put the following code block right after sector iterating loop:	

	for (u16_t i = 0; i < fs->sector_count; i++) {
		(....)
	}

	if (fs->sector_count == 2) {
		nvs_sector_advance(fs, &addr);

		rc = nvs_flash_cmp_const(fs, addr - ate_size, 0xff,
			sizeof(struct nvs_ate));
		if (!rc) {
			nvs_sector_advance(fs, &addr);
		}
	}

Laczen, 2019-05-20:
> Maybe, we should move it outside this loop so this doesn't affect performance when more than 2 sectors are defined? We can put the following code block right after sector iterating loop:
>
> ```
> for [...]

I think it's better to keep it in the loop. Even in the case where there are only 2 sectors you can detect a closed sector followed by an open sector and you would have jumped out the loop already with addr pointing to the correct sector. I made a correction above so that the check is only carried out for the last sector.

nvlsianpu, 2019-05-22:
fixed by #16280

closed 2019-05-22 by nvlsianpu