zephyrproject-rtos/zephyr #16215: FIFO queue data seems to get overwritten

jwknaup, 2019-05-16:
When implementing a fifo queue, do the data items need to be persistent until they are read out of the queue? 

In the below code, my output is 22, when I would expect it to be 12.

`struct data_item_t {
    unsigned int *fifo_reserved;   /* 1st word reserved for use by fifo */
    unsigned int foo;
};

struct data_item_t tx_data;
struct data_item_t *rx_data;

K_FIFO_DEFINE(my_fifo);

void put(int num){
    /* create data item to send */
    tx_data.foo = num;

    /* send data to consumers */
    k_fifo_put(&my_fifo, &tx_data);
}

void main(void)
{

    put(1);
    put(2);
    rx_data = k_fifo_get(&my_fifo, K_NO_WAIT);
    printk("%d", rx_data->foo);
    rx_data = k_fifo_get(&my_fifo, K_NO_WAIT);
    printk("%d", rx_data->foo);
}`

I am using v1.14.0 and building for a nrf52 board

jli157, 2019-05-16:
You need to create a new instance for a new data item in the function `put`.  Otherwise, there is only one instance in the queue and it is always overwritten by the next `put` operation.

jwknaup, 2019-05-16:
So like this?

```
struct data_item_t {
    unsigned int *fifo_reserved;   /* 1st word reserved for use by fifo */
    unsigned int foo;
};

K_FIFO_DEFINE(my_fifo);

void put(int num){

    struct data_item_t tx_data;
    /* create data item to send */
    tx_data.foo = num;

    /* send data to consumers */
    k_fifo_put(&my_fifo, &tx_data);
}

void get(){
    struct data_item_t *rx_data;

    rx_data = k_fifo_get(&my_fifo, K_NO_WAIT);
    printk("%d\n", rx_data->foo);
}

void main(void)
{

    put(1);
    put(2);
    get();
    get();

}
```

jli157, 2019-05-16:
The tx_data should not be a local variable since it will be invalid once the function `put` quits. You can use either malloc or set up a [mempool](https://docs.zephyrproject.org/latest/reference/kernel/memory/pools.html) to allocate a buffer for your new data item.

jukkar, 2019-05-16:
> struct data_item_t tx_data;

as @jli157 just commented, you cannot send stack variables via fifo.

alexanderwachter, 2019-05-17:
@jwknaup you can use a message queue instead (`k_msgq_*`).
https://docs.zephyrproject.org/latest/reference/kernel/data_passing/message_queues.html

aescolar, 2019-05-20:
It would seem this question has been answered => closing. If that is not the case, please reopen it.

closed 2019-05-20 by aescolar