zephyrproject-rtos/zephyr #16223: cdc_acm_composite sample does not work

frantony, 2019-05-17:
**Describe the bug**
cdc_acm_composite does not work on olimexino_stm32. Reverting commit e5cbe6a9e7 (usb: cdc: Add unique PIDs for each sample) fixes the problem.

 
**To Reproduce**
Steps to reproduce the behavior:

1. compile ```subsys/usb/cdc_acm_composite``` sample
```
cd samples/subsys/usb/cdc_acm_composite
rm -rf build && mkdir build && cd build
cmake -DBOARD=olimexino_stm32  ..
make
```
2. burn zephyr image into olimexino_stm32;
3. connect olimexino_stm32 to linux host via USB;
4. reset olimexino_stm32;
5. No ```/dev/ttyACMx``` device registered on linux host.
See ```dmesg``` output on linux host (some details may vary):
```
usb 2-1.1: new full-speed USB device number 102 using ehci-pci
usb 2-1.1: New USB device found, idVendor=2fe3, idProduct=0002, bcdDevice= 0.11
usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 2-1.1: can't set config #1, error -32
```

**Expected behavior**
Registering two new ```/dev/ttyACMx``` devices is expected on linux host.
Expected ```dmesg``` output on linux host (some details may vary):
```
usb 2-1.1: new full-speed USB device number 101 using ehci-pci
usb 2-1.1: New USB device found, idVendor=2fe3, idProduct=0100, bcdDevice= 0.11
usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 2-1.1: Product: Zephyr CDC ACM sample
usb 2-1.1: Manufacturer: ZEPHYR
usb 2-1.1: SerialNumber: 0.01
cdc_acm 2-1.1:1.0: ttyACM0: USB ACM device
cdc_acm 2-1.1:1.2: ttyACM1: USB ACM device
```

**Environment (please complete the following information):**

USB Host OS: Debian Linux x86_64, kernel version 4.18.6-1
Toolchain: Zephyr SDK 0.10.0
Zephyr git describe: zephyr-v1.14.0-712-g59eb48329e

erwango, 2019-05-17:
@masz-nordic, seems related to https://github.com/zephyrproject-rtos/zephyr/pull/14106

masz-nordic, 2019-05-17:
I do not have the `olimexino_stm32` board, so here is my setup:
Board: `nrf52840_pca10056`
OS: Ubuntu 16.04.1, kernel 4.15.0-48

Everything seems to work fine:
```
[  163.816690] usb 3-4: new full-speed USB device number 8 using xhci_hcd
[  163.967582] usb 3-4: New USB device found, idVendor=2fe3, idProduct=0002
[  163.967585] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  163.967588] usb 3-4: Product: Zephyr CDC ACM Composite sample
[  163.967590] usb 3-4: Manufacturer: ZEPHYR
[  163.967592] usb 3-4: SerialNumber: 0.01
[  163.968485] cdc_acm 3-4:1.0: ttyACM0: USB ACM device
[  163.969034] cdc_acm 3-4:1.2: ttyACM1: USB ACM device
[  168.039140] usb 3-4: USB disconnect, device number 8
[  168.476673] usb 3-4: new full-speed USB device number 9 using xhci_hcd
[  168.628028] usb 3-4: New USB device found, idVendor=2fe3, idProduct=0002
[  168.628030] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  168.628031] usb 3-4: Product: Zephyr CDC ACM Composite sample
[  168.628032] usb 3-4: Manufacturer: ZEPHYR
[  168.628032] usb 3-4: SerialNumber: 0.01
[  168.629054] cdc_acm 3-4:1.0: ttyACM0: USB ACM device
[  168.629544] cdc_acm 3-4:1.2: ttyACM1: USB ACM device
```

From your error `usb 2-1.1: can't set config #1, error -32` I would expect this has something to do with [problem I mentioned in other issue](https://github.com/zephyrproject-rtos/zephyr/issues/15917#issuecomment-492630853).

@frantony Can you try #16193 and see if it changes anything?

frantony, 2019-05-20:
@masz-nordic @finikorg 

> Can you try #16193 and see if it changes anything?

I have just tried b5aefbd from #16193 . Alas ```cdc_acm_composite```  does not work on ```olimexino_stm32```.

Also I have tested #16193 on ```olimex_stm32_e407``` (see #16264). The result is the same: ```cdc_acm_composite```  does not work, reverting commit e5cbe6a (usb: cdc: Add unique PIDs for each sample) fixes the problem.

finikorg, 2019-05-21:
I think real bug is with sending string descriptor > 64 bytes on stm32, I will check it

frantony, 2019-05-21:
@finikorg 

I have tested #16274 on ```olimex_stm32_e407``` (see #16264).

#16274 fixes #16223 with one drawback: ttyACMx registration time interval has been dramatically increased. 

Please see dmesg linux host output below.

The first one dmesg output: #16264 + revert e5cbe6a, the time between new USB device discovery and ttyACMx registration is less than **0.2s**:
```
[1730972.489077] usb 3-1.4.1: new full-speed USB device number 23 using ehci-platform
[1730972.642072] usb 3-1.4.1: New USB device found, idVendor=2fe3, idProduct=0100
[1730972.642092] usb 3-1.4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1730972.642104] usb 3-1.4.1: Product: Zephyr CDC ACM sample
[1730972.642116] usb 3-1.4.1: Manufacturer: ZEPHYR
[1730972.642127] usb 3-1.4.1: SerialNumber: 0.01
[1730972.643740] cdc_acm 3-1.4.1:1.0: ttyACM0: USB ACM device
[1730972.645637] cdc_acm 3-1.4.1:1.2: ttyACM1: USB ACM device
```

The second one dmesg output: #16264 + #16274, the time between new USB device discovery and ttyACMx registration is more than **5s**:
```
[1808756.810511] usb 3-1.4.1: new full-speed USB device number 24 using ehci-platform
[1808761.975487] usb 3-1.4.1: New USB device found, idVendor=2fe3, idProduct=0002
[1808761.975498] usb 3-1.4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1808761.975503] usb 3-1.4.1: Product: Zephyr CDC ACM Composite sample
[1808761.975508] usb 3-1.4.1: Manufacturer: ZEPHYR
[1808761.975513] usb 3-1.4.1: SerialNumber: 0.01
[1808761.976725] cdc_acm 3-1.4.1:1.0: ttyACM0: USB ACM device
[1808761.977995] cdc_acm 3-1.4.1:1.2: ttyACM1: USB ACM device
```

finikorg, 2019-05-22:
@frantony The real problem is with stm32 unable to send packet equal to 64 bytes to control endpoint (the string descriptor happens to be 64 bytes, if you change `USB Device Product` name by adding / removing character you get it "fixed"). My patch does not allow device to hang but does not fix stm32 issue. @ydamigos Could you check this?

ydamigos, 2019-05-23:
@finikorg @frantony I am able to see in wireshark the string descriptor reaching the host and then the host requesting it again. Maybe something is broken with ZLP in the driver. l'll debug it further.

finikorg, 2019-05-29:
> @finikorg @frantony I am able to see in wireshark the string descriptor reaching the host and then the host requesting it again. Maybe something is broken with ZLP in the driver. l'll debug it further.

carlescufi, 2019-05-29:
> @finikorg @frantony I am able to see in wireshark the string descriptor reaching the host and then the host requesting it again. Maybe something is broken with ZLP in the driver. l'll debug it further.

Thanks @ydamigos. A workaround PR has been posted by @finikorg here: https://github.com/zephyrproject-rtos/zephyr/pull/16274

This workaround would not be needed if you get to the bottom of the issue in the driver, so please let us know if we should merge the workaround or wait for  a fix from you.

ydamigos, 2019-05-29:
> so please let us know if we should merge the workaround or wait for a fix from you.

@carlescufi It depends how long you can wait. I work on Zephyr on my spare time and I cannot make any promises when it will be fixed.

carlescufi, 2019-06-05:
@ydamigos we will wait for now, please let us know when you have a patch for this.

ydamigos, 2019-07-02:
The issue is caused because the code runs in interrupt context and the write_sem (STM32 driver) is taken when usb_device tries to send the zero length packet without waiting for write done event, so it keeps retrying because IN callback is never called to give the write_sem back.
PR #17259 solves the issue.

commit 77125eaf (2019-09-09): usb: device: Allow limited amount of tries for write

Set limited amount of tries to write to USB control endpoint EP0.

Fixes #16223
closed 2019-09-09 by nashif