zephyrproject-rtos/zephyr #16224: tests/subsys/storage/flash_map meet mpu hardfault in trw_ke18f board.

hakehuang, 2019-05-17:
**Describe the bug**
when enable the flash_map test for twr_ke18f board, Ican see below message
```
starting test - test_flash_area_get_sectors
***** BUS FAULT *****
  Instruction bus error
  NXP MPU error, port 3
    Mode: Supervisor, Instruction Address: 0x20000810
    Type: Read, Master: 0, Regions: 0x8800
***** Hardware exception *****
Current thread ID = 0x20000880
Faulting instruction address = 0x20000810
Fatal fault in thread 0x20000880! Aborting.
```

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=twr_ke18f ..
3. make
4. See error

**Expected behavior**
the image 1 can be use  for testing.

**Impact**
falsh config error in twr_ke18f.dts for image_1
```
		slot1_partition: partition@3e000 {
			label = "image-1";
			reg = <0x0003e000 0x32000>;
		};
```

**Screenshots or console output**
```
starting test - test_flash_area_get_sectors
***** BUS FAULT *****
  Instruction bus error
  NXP MPU error, port 3
    Mode: Supervisor, Instruction Address: 0x20000810
    Type: Read, Master: 0, Regions: 0x8800
***** Hardware exception *****
Current thread ID = 0x20000880
Faulting instruction address = 0x20000810
Fatal fault in thread 0x20000880! Aborting.
```

**Environment (please complete the following information):**
 - OS: Linux,
 - Toolchain  Zephyr SDK
 - Commit SHA: c8708d9bf30bae2ccb2eab3d6c274107ddc36259

hakehuang, 2019-05-17:
@MaureenHelm

MaureenHelm, 2019-05-21:
@brix will you please take a look at this?

henrikbrixandersen, 2019-05-21:
So, it fails in `flash_erase()` for image-1 due to an MPU hardfault. What I do not understand is, that if I expand the size of MPU region 3 (the flash region) in `soc/arm/nxp_kinetis/ke1xf/nxp_mpu_regions.c` to cover from 0x0 to 0x7FFFFFFF, the test passes.

@MaureenHelm Any idea what I am doing wrong in the current NXP MPU region configuration?

henrikbrixandersen, 2019-05-26:
The problem arises from the MCUX SDK. In `fsl_ftfx_controller.c` it copies the contents of an array (`s_ftfxRunCommandFunctionCode`) to RAM (`s_ftfxRunCommand`) and calls it as a function (`config->runCmdFuncAddr`). This is not allowed by the MPU so a hardfault occurs

PR #16526 (2019-05-31): tests: flash_map: configure MPU to allow flash writes on twr_ke18f
Configure the NXP MPU to allow flash writes when running the flash_map
test.

This fixes #16224.
commit fd3d4156 (2019-05-31): tests: flash_map: configure MPU to allow flash writes on twr_ke18f

Configure the NXP MPU to allow flash writes when running the flash_map
test.

This fixes #16224.
closed 2019-05-31 by MaureenHelm