zephyrproject-rtos/zephyr #16225: tests/kernel/msgq/msgq_api twr_ke18f fails with assert

hakehuang, 2019-05-17:
**Describe the bug**
assert failure

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=twr_ke18f ..
3. make
4. See error

**Expected behavior**
PASS

**Impact**
message queue issue

**Screenshots or console output**
```
***** Booting Zephyr OS 1.14.99 *****
ASSERTION FAIL [num_parts <= max_partitions] @ ./kernel/mem_domain.c:90
***** Kernel Panic! *****
Current thread ID = 0x2000070c
Faulting instruction address = 0xa154
```
**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA: c8708d9

hakehuang, 2019-05-17:
@MaureenHelm

hakehuang, 2019-05-17:
tests/kernel/semaphore/sema_api
tests/kernel/semaphore/semaphore
tests/kernel/timer/timer_api
tests/kernel/workq/work_queue_api/
has the same failure

henrikbrixandersen, 2019-05-25:
These all fail due to the combination of limited number of MPU regions in the KE1xF SoC series and userspace, see the full explanation in #16229.

PR #18647 (2019-09-03): Test fix ke1xf userspace failures
Add an option to exlude HW Stack protection from user mode tests. This allows us to run the user mode test suite for platforms with limited number of MPU regions, such as the twr_ke18f board.
Fixes #18284 
Fixes #16229
Fixes #16225
closed 2019-09-03 by ioannisg