zephyrproject-rtos/zephyr #16231: Add CONFIG_UART_DYNAMIC_SETTINGS option

erwango, 2019-05-17:
**Is your enhancement proposal related to a problem? Please describe.**
Add an option to disable dynamic settings in uart.h (uart_configure, and uart_configure_get).
This saves around 500 bytes of flash per UART device (seen on STM32)

**Describe the solution you'd like**
Add a config UART_DYNAMIC_SETTINGS which would be conditional for uart settings related apis:
 - uart_configure
 - uart_configure_get
When disabled, they would not be available, and settings values would be the ones set at startup either from device tree or HW default values.

Default choice for this new value could be either n or y.
I'd propose n.

erwango, 2021-02-05:
@dcpleung I'm adding you as assignee. Feel free to close if you think this is not something we want.

dcpleung, 2021-02-05:
Maybe something more like `CONFIG_UART_SUPPORT_RUNTIME_CONFIGURE`? I just looked at a couple drivers and they call configure() during init(), and `uart_apbuart` doesn't configure the port during init (... possibly relying on runtime config). So I think in general this doesn't save us much in code size on other platforms.

Is it something you can do in the STM32 driver? If you set `.configure` and `.config_get` to null in API struct, and ifdef the functions, it would archive some code saving. Calling to API would simply return -ENOTSUP in this case.

PR #35697 (2021-06-07): serial: introduce CONFIG_UART_USE_RUNTIME_CONFIGURE
This kconfig option enables runtime configuration of UART
controllers. This allows application to call uart_configure()
to configure the UART controllers and calling uart_config_get()
to retrieve configuration. If this is disabled, UART controllers
rely on UART driver's initialization function to properly
configure the controller. The main use of this option is mainly
code size reduction.

Fixes #16231
commit 4e1692f8 (2021-06-07): serial: introduce CONFIG_UART_USE_RUNTIME_CONFIGURE

This kconfig option enables runtime configuration of UART
controllers. This allows application to call uart_configure()
to configure the UART controllers and calling uart_config_get()
to retrieve configuration. If this is disabled, UART controllers
rely on UART driver's initialization function to properly
configure the controller. The main use of this option is mainly
code size reduction.

Fixes #16231
closed 2021-06-07 by carlescufi