zephyrproject-rtos/zephyr #16232: STM32: implement pinmux api

erwango, 2019-05-17:
**Is your enhancement proposal related to a problem? Please describe.**
pinmux.h is not implemented for STM32 targets

**Describe the solution you'd like**
Implement pinmux API for stm32 series

psidhu, 2019-05-17:
Quick question: Are we going to ever move to pinmux in dts?

erwango, 2019-05-20:
@psidhu: Yeah, but from my view, this is likely to be the last piece converted to dts.
Reason is I see 2 prerequisites that are not there yet:
- To be easy to use, it will require that we're able to generate code from dts (at least: not only #define 's). 
- We need a Linux like pinctrl API, used widely, so we only have one API to focus on for dts > code transition.

erwango, 2020-02-11:
Closing this point as we're likely focus efforts on implementing DT related pinmux/pinctrl API

closed 2020-02-11 by erwango