zephyrproject-rtos/zephyr #16234: tests/benchmarks/latency_measure can not meet real time requirement for twr_ke18f

hakehuang, 2019-05-17:
**Describe the bug**
tests/benchmarks/latency_measure can not meet real time requirement for twr_ke18f

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=twr_ke18f ..
3. make
4. See error

**Expected behavior**
test pass

**Impact**
real time schedule for tasks

**Screenshots or console output**
```
***** Booting Zephyr OS 1.14.99 *****

|-----------------------------------------------------------------------------|

|                            Latency Benchmark                                |

|-----------------------------------------------------------------------------|

|  tcs = timer clock cycles: 1 tcs is 8 nsec                                  |

|-----------------------------------------------------------------------------|

| 1 - Measure time to switch from ISR back to interrupted thread              |

| switching time is 870 tcs = 7250 nsec                                       |

|-----------------------------------------------------------------------------|

| 2 - Measure time from ISR to executing a different thread (rescheduled)     |

| switch time is 1187 tcs = 9891 nsec                                         |

|-----------------------------------------------------------------------------|

| 3 - Measure average time to signal a sema then test that sema               |

| Average semaphore signal time 802 tcs = 6683 nsec                           |

| Average semaphore test time 92 tcs = 766 nsec                               |

|-----------------------------------------------------------------------------|

| 4- Measure average time to lock a mutex then unlock that mutex              |

| Average time to lock the mutex 875 tcs = 7291 nsec                          |

| Average time to unlock the mutex 118268 tcs = 985568 nsec                   |

|-----------------------------------------------------------------------------|

| 5 - Measure average context switch time between threads using (k_yield)     |

| Average thread context switch using yield 1067 tcs = 8896 nsec              |

|-----------------------------------------------------------------------------|

| 6 - Measure average context switch time between threads (coop)              |

| Error: tick occurred                                                        |

|-----------------------------------------------------------------------------|

===================================================================

PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: ( Linux)
 - Toolchain ( Zephyr SDK, ...)
 - Commit SHA c8708d9

hakehuang, 2019-05-17:
@MaureenHelm

henrikbrixandersen, 2019-05-21:
It works when doing `cmake -DBOARD=twr_ke18f -DCONF_FILE=prj_small_freq_divider.conf ..`.

hakehuang, 2019-05-22:
@hackermnementh this configure works. thanks

PR #16394 (2019-05-31): tests: benchmarks: latency_measure: use 0.1s sys clk tick for twr_ke18f
Use the small_freq_divider.conf file when running on the NXP TWR-KE18F
development board.

Fixes #16234.
commit 09749bbe (2019-05-31): tests: benchmarks: latency_measure: use 0.1s sys clk tick for twr_ke18f

Use a system clock tick of 0.1 sec when running on the NXP TWR-KE18F
development board.

Fixes #16234.
closed 2019-05-31 by MaureenHelm