zephyrproject-rtos/zephyr #16236: [docs] Windows installation guide, git part, is installed with non-intended configuration

n3rd4i, 2019-05-17:
**Describe the bug**
If one decides to follow the **windows installation guide #1** from zephyr wiki you will later have an issue with git. Default git-for-windows installations breaks line endings. It converts them from LF to CRLF. Because of this the project will break when trying to source the *.sh environment (from WSL or VM) with this **zephyrproject** installation.

**To Reproduce**
Steps to reproduce the behavior:
1. follow Option 1: Windows Command Prompt, ending up with:
```
choco install git python ninja dtc-msys2 gperf
```
2. source `zephyr-env.sh` from `WSL` or `shared-VM-folder`
3. See error:
```log
-bash: $'\r': command not found
-bash: zephyr-env.sh: line 70: syntax error: unexpected end of file
```

**Expected behavior**
Shell script should be sourced successfully. 

**Impact**
For people which don't know the issue could be: `showstopper` for people which understand the issue could be just `annoyance`.

**Environment (please complete the following information):**
 - OS: Windows [10]
 - Toolchain: Zephyr SDK
 - Version: 1.14.99 (docs release)

**Additional context**
Proposed solution is to change the installation instructions from:
Before: `choco install git python ninja dtc-msys2 gperf`
After: 
```
choco install python ninja dtc-msys2 gperf
choco install git.install --params "/NoAutoCrlf"
```

carlescufi, 2019-05-17:
This is not a bug, because for WSL or a VM we clearly state to follow another set of instructions, which will install the native git client for those:
https://docs.zephyrproject.org/latest/getting_started/installation_win.html#option-2-windows-10-wsl-windows-subsystem-for-linux

>Follow the Ubuntu instructions in the Development Environment Setup on Linux document.

carlescufi, 2019-05-17:
In general it is not a good idea to share a tree between `cmd.exe` and WSL or a VM, because of these type of issues (there are others, like file permissions). I recommend you have separate Git trees, each cloned with the appropriate `git` executable

andrewboie, 2019-05-17:
Yeah this doesn't seem right, you shouldn't be mixing WSL and native Windows applications, you will run into all kinds of issues.

carlescufi, 2019-05-21:
Closing this as it's already documented as intended

closed 2019-05-21 by carlescufi