zephyrproject-rtos/zephyr #16237: disco_l475_iot1: samples/bluetooth/ipsp ko since 3151d265726fb528d31d8e1ee8b35004e901cfc4

erwango, 2019-05-17:
**Describe the bug**
`samples/bluetooth/ipsp` doesn't work anymore on disco_l475_iot1 since merge of 3151d265726fb528d31d8e1ee8b35004e901cfc4

**To Reproduce**
Go before 3151d265726fb528d31d8e1ee8b35004e901cfc4
Test `samples/bluetooth/ipsp` (using CONFIG_NET_L2_BT_ZEP1656=y)
It is possible to ping the device from a remote
Go after 3151d265726fb528d31d8e1ee8b35004e901cfc4
Test `samples/bluetooth/ipsp` (using CONFIG_NET_L2_BT_ZEP1656=y)
No more response to ping
**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
```
root@stih410-b2260:~# echo "connect c8:2e:16:88:95:8e 2" > /sys/kernel/debug/bluetooth/6lowpan_control                     
root@stih410-b2260:~# ip address add 2001:db8::2/64 dev bt0
root@stih410-b2260:~# ping6 2001:db8::1
PING 2001:db8::1(2001:db8::1) 56 data bytes
ping: sendmsg: No buffer space available
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain (e.g Zephyr SDK 0.9.5)
 - Commit SHA: 3151d265726fb528d31d8e1ee8b35004e901cfc4

**Additional context**
Add any other context about the problem here.

erwango, 2019-05-17:
@Vudentz , @jhedberg, this is more fyi than a real bug I'd like to see solved quickly.
This regression seems limited to disco_l475_iot1 (due to the fact it is BT4.1?).

Anyway, I've done a bit of archeology to find out the exact root cause of the regression, so I prefer to log it so information is not lost.
This use case is not a priority for me now, so I don't have much more time to spend on it, but in case it rings a bell to you, I would welcome any suggestion.

nashif, 2020-03-11:
@jhedberg , @erwango  is this still relevant?

erwango, 2020-03-12:
@nashif I was planning to test this in coming weeks.
So if no urgency, let's keep it open few weeks more.

closed 2020-10-04 by github-actions[bot]