zephyrproject-rtos/zephyr #16238: k_cycle_get_32() API is useless in some Zephyr configurations

pizi-nordic, 2019-05-17:
TL;DR:
`CONFIG_USERSPACE=y` + `CONFIG_TIMER_READS_ITS_FREQUENCY_AT_RUNTIME=y` = No reference for k_cycle_get_32().

The `CONFIG_USERSPACE` option introduces strong separation between the application and the 
kernel memory. The `CONFIG_TIMER_READS_ITS_FREQUENCY_AT_RUNTIME` makes the `CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC` define useless leaving the `sys_clock_hw_cycles_per_sec()` function the only source of system timer frequency, which is needed to interpret data from k_cycle_get_32().

Unfortunately, the `sys_clock_hw_cycles_per_sec()` function directly reads the kernel memory (z_clock_hw_cycles_per_sec) triggering exception when called from user thread.

pizi-nordic, 2019-05-17:
Currently this bug, together with https://github.com/zephyrproject-rtos/zephyr/pull/15230 (which prevents the tests from **not using** userspace if platform supports it), blocks https://github.com/zephyrproject-rtos/zephyr/pull/16203,  which is a fix for https://github.com/zephyrproject-rtos/zephyr/issues/15983.

pizi-nordic, 2019-05-17:
FYI: @nashif, @andrewboie, @carlescufi

andrewboie, 2019-05-21:
what I think is needed is a new, private z_ prefixed API, like z_sys_runtime_clock_hw_cycles_per_sec() which returns runtime calculated system clock frequency. make it a system call. and then wrap sys_clock_hw_cycles_per_sec to either call that or just return CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC

pizi-nordic, 2019-05-21:
I would rather name it k_cycle_freq_get() in order to relate it to k_cycle_get_32(). IMHO on platforms witch clock speed known at compile time, it could be a static inline returning constant value and syscall on the others. What do you think?

andrewboie, 2019-05-21:
> it could be a static inline returning constant value and syscall on the others. What do you think?

I don't care what you call it, but you can't ifdef its definition.
The script that scans for syscall names in header files doesn't use the preprocessor.

pizi-nordic, 2019-05-22:
>I don't care what you call it, but you can't ifdef its definition.
The script that scans for syscall names in header files doesn't use the preprocessor. 

Ok. I have not known that.

PR #16302 (2019-05-22): kernel: timeout: add syscall for runtime clk freq
If the system sets its clock frequency at runtime, this is
stored in a variable that can't be directly read by user
mode. For this case only, add a system call to fetch its
value and modify the definition of
sys_clock_hw_cycles_per_sec() to use it.

Since this is now a system call, store in a temporary variable
inside z_ms_to_ticks(). The syscall overhead only applies
when called from user mode, other contexts are completely
inlined.

Fixes: #16238
commit fd49cf7d (2019-05-22): kernel: timeout: add syscall for runtime clk freq

If the system sets its clock frequency at runtime, this is
stored in a variable that can't be directly read by user
mode. For this case only, add a system call to fetch its
value and modify the definition of
sys_clock_hw_cycles_per_sec() to use it.

Since this is now a system call, store in a temporary variable
inside z_ms_to_ticks(). The syscall overhead only applies
when called from user mode, other contexts are completely
inlined.

Added stub syscall header for mocking framework, to get rid
of inclusion errors.

Fixes: #16238
closed 2019-05-22 by carlescufi