zephyrproject-rtos/zephyr #16240: USB Bluetooth and DFU classes cannot be enabled simultaneously on nRF52840

AdamMooers, 2019-05-17:
**Describe the bug**
Whenever the USB DFU class (`USB_DFU_CLASS`) and the USB Bluetooth class (`USB_DEVICE_BLUETOOTH`) are both enabled, neither works correctly. I have tested DFU and Bluetooth separately and both work. DFU also works when the bluetooth kernel module is removed in Linux (via modprobe btusb).

**Configuration (prj.conf)**
```
CONFIG_STDOUT_CONSOLE=y
CONFIG_GPIO=y
CONFIG_SERIAL=y
CONFIG_UART_INTERRUPT_DRIVEN=y

CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC=y
CONFIG_CLOCK_CONTROL_NRF_K32SRC_20PPM=y

CONFIG_BT=y
CONFIG_BT_HCI_RAW=y
CONFIG_BT_CTLR_TX_BUFFER_SIZE=251
CONFIG_BT_RX_BUF_LEN=258
CONFIG_BT_CTLR_DATA_LENGTH_MAX=251

CONFIG_USB=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DFU_CLASS=y
CONFIG_USB_DEVICE_BLUETOOTH=y

CONFIG_FLASH=y
CONFIG_SOC_FLASH_NRF=y
CONFIG_BOOTLOADER_MCUBOOT=y
CONFIG_IMG_MANAGER=y
CONFIG_MCUBOOT_IMG_MANAGER=y

CONFIG_USB_DEVICE_VID=0x167f
CONFIG_USB_DEVICE_PID=0x3001

CONFIG_USB_DEVICE_MANUFACTURER="Company"
CONFIG_USB_DEVICE_PRODUCT="Product"
CONFIG_USB_DEVICE_SN="123abc"
```
**To Reproduce**
- source zephyr/zephyr/zephyr-env.sh
- mkdir -p build/nrf52840_pca10056 && cd build/nrf52840_pca10056
- cmake -GNinja -DBOARD=nrf52840_pca10056 -Dapp_VERSION_BUILD=9999 ../..
- ninja
- ~/repos/mcuboot/scripts/imgtool.py sign --key ~/repos/mcuboot/root-ec-p256.pem --header-size 0x200 --align 8 --version 1.2 -S 0x69000 ./zephyr/zephyr.hex signed-hello.hex
- nrfjprog --family NRF52 --program signed-hello.hex --sectorerase
- ~/repos/mcuboot/scripts/imgtool.py sign --key ~/repos/mcuboot/root-ec-p256.pem --header-size 0x200 --align 8 --version 1.2 -S 0x69000 ./zephyr/zephyr.bin signed-hello.bin
- sudo dfu-util --alt 1 --download signed-hello.bin

**Expected behavior**

- sudo dfu-util --alt 1 --download signed-hello.bin
```
dfu-util 0.9

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 167f:3001
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #1 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 64
Copying data from PC to DFU device
Download    [=========================] 100%        85970 bytes
Download done.
state(2) = dfuIDLE, status(0) = No error condition is present
Done!
```

- hciconfig
```
hci1:   Type: Primary  Bus: USB
        BD Address: 94:54:93:36:44:B8  ACL MTU: 251:7  SCO MTU: 0:0
        UP RUNNING
        RX bytes:233 acl:0 sco:0 events:19 errors:0
        TX bytes:120 acl:0 sco:0 commands:19 errors:0
```

**Actual Behaviour**
- sudo dfu-util --alt 1 --download signed-hello.bin
```
dfu-util 0.9

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 167f:3001
Run-time device DFU version 0110
Claiming USB DFU Runtime Interface...
dfu-util: Cannot claim interface 1: LIBUSB_ERROR_BUSY
```
- hciconfig
```
hci1:   Type: Primary  Bus: USB
        BD Address: 00:00:00:00:00:00  ACL MTU: 0:0  SCO MTU: 0:0
        DOWN INIT RUNNING 
        RX bytes:0 acl:0 sco:0 events:0 errors:0
        TX bytes:3 acl:0 sco:0 commands:1 errors:0
```
followed by
```
hci1:   Type: Primary  Bus: USB
        BD Address: 00:00:00:00:00:00  ACL MTU: 0:0  SCO MTU: 0:0
        DOWN 
        RX bytes:0 acl:0 sco:0 events:0 errors:0
        TX bytes:3 acl:0 sco:0 commands:1 errors:0
```

- hciconfig hci1 up
`Can't init device hci1: Connection timed out (110)`

**Impact**
DFU cannot be used in applications which require Bluetooth HCI over USB.

**Environment:**
- Ubuntu 16.04 (host)
- nRF52840 PCA10056 hw v. 1.1.0
- Zephyr 1.14.0 (`cebe115`)
- Zephyr SDK 0.10.0
- dfu-util 0.9 (modified for increased timeout during flash erase)
- MCUboot (master, `704b8ba`)

jfischer-phytec-iot, 2019-05-20:
@AdamMooers you need two patches below to get it working stable on Zephyr 1.14.0 (cebe115)

https://github.com/zephyrproject-rtos/zephyr/pull/15747
https://github.com/zephyrproject-rtos/zephyr/pull/14662

AdamMooers, 2019-05-21:
@jfischer-phytec-iot I pulled down master (74f0106) today and enabled `IMG_ERASE_PROGRESSIVELY`. I left `DFU_DEFAULT_POLLTIMEOUT` at its 256 ms default. Unfortunately, it appears like the same thing is happening: The DFU class works if the BT class is disabled via `prj.conf` and vice versa. 

Do I need to use an alternative branch?

jfischer-phytec-iot, 2019-05-22:
@AdamMooers Unfortunately, the combination hci_usb + usb_dfu does not work on master and 1.14.0 (with or without patches). Also combination hci_usb + msd did not work for me. There is ongoing work on compostite mode (#15818), after it is merged I will look for this issue again.

AdamMooers, 2019-05-22:
@jfischer-phytec-iot Thank you for looking into this issue and for the information about the work on composite mode. I will be sure to keep an eye on future developments.

jfischer-phytec-iot, 2019-06-26:
@AdamMooers There is a fix for this issue, but it is not possible (fortunately) to start a download or upload as long the device is claimed by Bluez and btusb module.

PR #17089 (2019-06-27): usb: sort usb config data by section
USBD_CFG_DATA_DEFINE macro has not consider that a class
could have more than one set of usb_cfg_data struct.
If a class has more than one set of usb_cfg_data
then they should be sorted the same way like by
USBD_DEVICE_DESCR_DEFINE macro.

Fixes: #16240
commit abaeaed2 (2019-06-27): usb: sort usb config data by section

USBD_CFG_DATA_DEFINE macro has not consider that a class
could have more than one set of usb_cfg_data struct.
If a class has more than one set of usb_cfg_data
then they should be sorted the same way like by
USBD_DEVICE_DESCR_DEFINE macro.

Fixes: #16240
closed 2019-06-27 by carlescufi