zephyrproject-rtos/zephyr #16243: std::vector push_back() not working correctly

zayralobo, 2019-05-17:
**Describe the bug**
When I use push_back() on a std::vector, I get incorrect values pushed back into my vector. The first value is consistently pushed back incorrectly, while the next few values seem to consistently be correct and then the later values are incorrect again. I get similar results when running on either a SAM E70 Xplained or ST Nucleo L476RG.

**To Reproduce**
prj.conf: 
~~~
CONFIG_CPLUSCPLUS=y
~~~

main.cpp: 
~~~
#include <zephyr.h>
#include <sensor.h>
#include <misc/printk.h>
#include <device.h>
#include <vector>
#include <stdio.h>
#include <stdlib.h>

int main() {
	std::vector<int> a;

	for (int i = 0; i < 100; i++) {
		a.push_back(i);
                printf("current size of vector: %d\n", a.size());
	}
	for (int i = 0; i < 100; i++) {
		printf("iter=%d value=%d\n", i, a[i]);	
	}

	return 0;
} 
~~~

CMakeLists.txt:
~~~
cmake_minimum_required(VERSION 3.13.1)

set(CMAKE_CXX_FLAGS "-fpermissive -std=c++0x")

macro(set_conf_file)
  if(EXISTS                 ${APPLICATION_SOURCE_DIR}/boards/${BOARD}.conf)
    set(CONF_FILE "prj.conf ${APPLICATION_SOURCE_DIR}/boards/${BOARD}.conf")
  else()
    set(CONF_FILE "prj.conf")
  endif()
endmacro()

include($ENV{ZEPHYR_BASE}/cmake/app/boilerplate.cmake NO_POLICY_SCOPE)
project(hs1_porting)

FILE(GLOB app_sources src/main.cpp)
target_sources(app PRIVATE
  ${app_sources}
  )

include($ENV{ZEPHYR_BASE}/samples/net/common/common.cmake)
~~~

1. Put prj.conf, CMakeLists.txt, and a folder called src in the samples folder of your zephyrproject. Put main.cpp in the src folder. 
2. mkdir build && cd build
3. cmake -DBOARD=sam_e70_xplained ..; sudo make -j16 flash

Output:
~~~
***** Booting Zephyr OS 1.14.0-rc3 *****
current size of vector: 1
current size of vector: 2
current size of vector: 3
current size of vector: 4
current size of vector: 5
current size of vector: 6
current size of vector: 7
current size of vector: 8
current size of vector: 9
current size of vector: 10
current size of vector: 11
current size of vector: 12
current size of vector: 13
current size of vector: 14
current size of vector: 15
current size of vector: 16
current size of vector: 17
current size of vector: 18
current size of vector: 19
current size of vector: 20
current size of vector: 21
current size of vector: 22
current size of vector: 23
current size of vector: 24
current size of vector: 25
current size of vector: 26
current size of vector: 27
current size of vector: 28
current size of vector: 29
current size of vector: 30
current size of vector: 31
current size of vector: 32
current size of vector: 33
current size of vector: 34
current size of vector: 35
current size of vector: 36
current size of vector: 37
current size of vector: 38
current size of vector: 39
current size of vector: 40
current size of vector: 41
current size of vector: 42
current size of vector: 43
current size of vector: 44
current size of vector: 45
current size of vector: 46
current size of vector: 47
current size of vector: 48
current size of vector: 49
current size of vector: 50
current size of vector: 51
current size of vector: 52
current size of vector: 53
current size of vector: 54
current size of vector: 55
current size of vector: 56
current size of vector: 57
current size of vector: 58
current size of vector: 59
current size of vector: 60
current size of vector: 61
current size of vector: 62
current size of vector: 63
current size of vector: 64
current size of vector: 65
current size of vector: 66
current size of vector: 67
current size of vector: 68
current size of vector: 69
current size of vector: 70
current size of vector: 71
current size of vector: 72
current size of vector: 73
current size of vector: 74
current size of vector: 75
current size of vector: 76
current size of vector: 77
current size of vector: 78
current size of vector: 79
current size of vector: 80
current size of vector: 81
current size of vector: 82
current size of vector: 83
current size of vector: 84
current size of vector: 85
current size of vector: 86
current size of vector: 87
current size of vector: 88
current size of vector: 89
current size of vector: 90
current size of vector: 91
current size of vector: 92
current size of vector: 93
current size of vector: 94
current size of vector: 95
current size of vector: 96
current size of vector: 97
current size of vector: 98
current size of vector: 99
current size of vector: 100
iter=0 value=541066560
iter=1 value=1
iter=2 value=2
iter=3 value=3
iter=4 value=4
iter=5 value=5
iter=6 value=6
iter=7 value=7
iter=8 value=4197129
iter=9 value=4197129
iter=10 value=4197129
iter=11 value=4196241
iter=12 value=4197129
iter=13 value=4197129
iter=14 value=4196153
iter=15 value=4195709
iter=16 value=4197433
iter=17 value=4197433
iter=18 value=4197433
iter=19 value=4197433
iter=20 value=4197433
iter=21 value=4197433
iter=22 value=4197433
iter=23 value=4197433
iter=24 value=4197433
iter=25 value=4197433
iter=26 value=4197433
iter=27 value=4197433
iter=28 value=4197433
iter=29 value=4197433
iter=30 value=4197433
iter=31 value=4197433
iter=32 value=4197433
iter=33 value=4197433
iter=34 value=4197433
iter=35 value=4197433
iter=36 value=4197433
iter=37 value=4197433
iter=38 value=4197433
iter=39 value=4197433
iter=40 value=4197433
iter=41 value=4197433
iter=42 value=4197433
iter=43 value=4197433
iter=44 value=4197433
iter=45 value=4197433
iter=46 value=4197433
iter=47 value=4197433
iter=48 value=4197433
iter=49 value=4197433
iter=50 value=4197433
iter=51 value=4197433
iter=52 value=4197433
iter=53 value=4197433
iter=54 value=4197433
iter=55 value=4197433
iter=56 value=4197433
iter=57 value=4197433
iter=58 value=4197433
iter=59 value=4197433
iter=60 value=4197433
iter=61 value=4197433
iter=62 value=4197433
iter=63 value=4197433
iter=64 value=4197433
iter=65 value=4197433
iter=66 value=4197433
iter=67 value=4197433
iter=68 value=4197433
iter=69 value=4197433
iter=70 value=4197433
iter=71 value=4197433
iter=72 value=4197433
iter=73 value=4197433
iter=74 value=4197433
iter=75 value=4197433
iter=76 value=4197433
iter=77 value=4197433
iter=78 value=4197433
iter=79 value=4197433
iter=80 value=4197433
iter=81 value=4197433
iter=82 value=4197433
iter=83 value=4197433
iter=84 value=4197433
iter=85 value=4197433
iter=86 value=4197433
iter=87 value=587248944
iter=88 value=1276752005
iter=89 value=-1828678909
iter=90 value=855763405
iter=91 value=727948032
iter=92 value=-371336169
iter=93 value=1116803586
iter=94 value=-1325281266
iter=95 value=822370315
iter=96 value=1176539394
iter=97 value=822208989
iter=98 value=277420745
iter=99 value=-99422207
~~~

**Expected behavior**
Each printed iter value should match the value pushed back and then printed. Currently this expected behavior is only occurring for the 2nd-8th values.

**Impact**
Have to hard code (i.e. a[0] = 0) each value I want in my vector after using push_back() to allocate space for it.

**Environment:**
 - OS: Ubuntu 16.04
 - Toolchain: Zephyr SDK 0.10.0
 - Version used: Zephyr 1.14.0-rc3

**Additional Information**
I tried the patch suggested in https://github.com/zephyrproject-rtos/zephyr/issues/15603 (modifying my target_baremetal.cmake) but I get the same incorrect results.

pabigot, 2019-05-21:
The values I see being displayed appear to be RAM addresses rather than the pushed value.

I can confirm that this problem is still present when #15603 is avoided by using `CONFIG_NEWLIB_LIBC=y` and when `ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb` is set to use the stock [gcc-arm-none-eabi-8-2018-q4-major](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads), which supports STL just fine.

So this is definitely something Zephyr is breaking in some subtle way.  Looking at the verbose build, I suspected the use of `-nostdlib` via `CMAKE_REQUIRED_FLAGS`, but removing that from `build.ninja` didn't help.

pabigot, 2019-05-21:
Resolved: Adding `CONFIG_HEAP_MEM_POOL_SIZE=8192` (which defaults to zero) fixes it.

So probably the code caused an exception to be thrown and ignored.

closed 2019-05-21 by pabigot