zephyrproject-rtos/zephyr #16273: Calling k_work_submit can reenable interrupts

pdunaj, 2019-05-21:
**Describe the bug**
We were hit by a recursive spinlock assert.
```
ASSERTION FAIL [z_spin_lock_valid(l)] @ ./include/spinlock.h:78
--
Recursive spinlock
```

Quick investigation has shown that it hits in unexpected place and the only possible explanation is that interrupts are being re-enabled while spinlock is held.
The code snippet is as follows:
```
funcA(...)
{
   spin_lock(); // this is where the lock is taken
   enable_gpio_callbacks();
   k_work_submit(work);
   spin_unlock(); // we don't get here before irq is fired
}

gpio_irq_handler()
{
   spin_lock(); // fired irq fails on recursive spinlock
   disbale_gpio_callbacks();
   spin_unlock();
}
```

Further investigation shown that `k_work_submit` can cause reschedule which can take CPU away from the `funcA` and cause interrupts to be re-enabled.

```
--- a/kernel/queue.c
+++ b/kernel/queue.c
@@ -181,6 +181,7 @@ static s32_t queue_insert(struct k_queue *queue, void *prev, void *data,
        handle_poll_events(queue, K_POLL_STATE_DATA_AVAILABLE);
 #endif /* CONFIG_POLL */
 
+       __ASSERT(key.key == 0, "Heh...");
        z_reschedule(&queue->lock, key);
        return 0;
 }
```

**To Reproduce**
Steps to reproduce the behavior:
1. See description

**Expected behavior**
We must be able to submit work in the atomic context. It should not sleep or reschedule at that point.

**Impact**
Showstopper

**Screenshots or console output**
N/A

**Environment (please complete the following information):**
master

**Additional context**
N/A

pdunaj, 2019-05-21:
Following seems to fix the bug:
```
--- a/kernel/queue.c
+++ b/kernel/queue.c
@@ -181,7 +181,11 @@ static s32_t queue_insert(struct k_queue *queue, void *prev, void *data,
        handle_poll_events(queue, K_POLL_STATE_DATA_AVAILABLE);
 #endif /* CONFIG_POLL */
 
-       z_reschedule(&queue->lock, key);
+       if (key.key == 0) {
+               z_reschedule(&queue->lock, key);
+       } else {
+               k_spin_unlock(&queue->lock, key);
+       }
        return 0;
 }
```

pdunaj, 2019-05-21:
@pizi-nordic

ioannisg, 2019-05-21:
@andyross FYI

pdunaj, 2019-05-21:
The same issue is within `z_impl_k_wakeup` which causes similar issue e.g. when calling `LOG_ERR` under the spinlock. You can see that it check if we are in isr before calling `z_reschedule_unlocked` but is not checking if interrupts are disabled. This will cause exactly the same issue.

Personally I believe that all places that can be run under isr should not call reschedule at all.
Also if we call any API causes scheduling this should lead to oops. This is the behavior in Linux as this is the only way we can be sure nobody does anything stupid.

andrewboie, 2019-05-21:
> Personally I believe that all places that can be run under isr or with irqs disabled (i.e. under atomic context) should not call reschedule at all.

We need some way to trigger rescheduling, for example if an ISR gives a semaphore, on the way out of the ISR if a high-priority thread (higher than the currently running one) was waiting on that semaphore, it needs to be scheduled.

andyross, 2019-05-21:
Yeah, that's a bug.  Note that this has never really done what you want: k_work_submit() (and anything else that calls z_reschedule()) has always acted as a schedule point and released your lock (and thus probably broken your synchronization).  It used to be nonfatal though, because irq_locks() are implicitly recursive (i.e. you can lock more than once) where spinlocks are not.  Now, you have to be holding exactly one lock at the point of z_swap().  And we do test for this and panic, which is exactly what that recursive spinlock assert was doing.

As for the fix above: it's the right idea I think (because queues should treat atomic contexts just like interrupt context and not block), but that code won't work for production because an irq_lock "key" value is architecture defined and not guaranteed to be zero for the "interrupts enabled" case.  What we really need is a new API like `z_arch_is_atomic()` or something, I guess.

andyross, 2019-05-21:
And @andrewboie : that's the way it works already.  On interrupt exit the scheduler chooses whoever the highest priority thread is.  But for obvious reasons you can't actually do the context switch in the interrupt context.

rakons, 2019-05-22:
Hi. I did not analyse how it is done in Zephyr, but on ARM, to make context switching safe it usually uses PendSV interrupt for that - then you do not have to check if interrupts are already blocked or not, you do not have to switch context by manual call inside functions that need that and finally, you would get context switch if needed just after interrupts are unlocked.

pizi-nordic, 2019-05-22:
I think that this problem is much wider. There is a plenty of APIs which are calling z_reschedule() and friends, without checking if there is no spin/irq lock held in layers above. This issue raises a big question about overall stability of many existing and future subsystems. Especially, that the lock could be held several layers above the Zephyr API call.

I just did a short search in the code, and found the following examples (easy to identify as irq lock is taken on the same level as k_* API call): 

eth_stellaris_isr() -> k_sem_give() called with irqs locked.
eth_mcux_dispacher_isr() -> k_work_submit() called with irqs locked.
usb_transfer() -> k_work_submit() called with irqs locked.

IMHO the real problem is that nobody knows what else is affected as tracing the problem in the code is hard. For example: Could you tell at first glance that the interrupts could be re-enabled in usb_dw_tx() in the middle of the function when data_len is larger than max_xfer_size?

andyross, 2019-05-22:
That's handled at the API layer, essentially.  The spinlockification work split the reschedule call into a few variants.  There is a z_reschedule_unlocked() call used in lots of spots where "I might have woken someone up and they might need to run".   This is the one that should be checking for an atomic (and not just ISR) context.  It's very reasonable that someone up the call stack might have their own synchronization live.

The other ones take an irqlock or a spinlock key, and they're used in circumstances where: "I've atomically decided that someone else must run so do that right now, releasing this lock at the same time".   In this circumstances, you **need** to be sure no one else has a lock active or else you'll break **their** synchronization (e.g. by creating a deadlock by going to sleep while holding a resource someone else is waiting on, etc...).  For these, the existing error checking is I think appropriate (though it could surely be cleaned up some).

It's also worth pointing out that these aren't widely used APIs.  They're kernel-internal scheduling gadgets.  Applications and drivers should be writing to the more conventionally defined IPC layer.  In this case the k_queue code can very reasonably rely on a z_reschedule_unlocked() that worked as described above.

andyross, 2019-05-22:
And again: this isn't a new bug, it's been our behavior forever.  It's just that before we'd silently swap anyway while holding that lock, where now we detect it as an error.

andyross, 2019-05-22:
@rakons: We do indeed use PendSV on ARM.  But that has its own gotchas.  Check the tree for instances of CONFIG_SWAP_NONATOMIC where we have to work around the resulting craziness.  Personally I don't think this is a win at all.  In any case Zephyr never actually context switches in interrupt context -- we have a separate interrupt stack so we can't suspend it.  Threads state gets saved at interrupt entry (sorta, there are lots of arch-specific optimizations in this space) such that any thread can be restored at exit (or via z_swap() from thread mode).

carlescufi, 2019-05-22:
@andyross `k_work_submit` is used extensively in our downstream, which provides additional subsystems to the one upstream, so this is a critical bug in this context. Is the new required API you mention, `z_arch_is_atomic` something you could draft?

pabigot, 2019-05-23:
AFAICT `k_poll_signal_raise()` has a similar behavior, in that it invokes `z_reschedule()` with the kernel spin lock that it holds, disregarding anything that might be held in the caller.  So the test with `z_arch_is_atomic()` might need to be added there too.  Assuming that test can detect that other spinlocks are also being held.

Or maybe this should be done in `z_reschedule()` where it should solve all cases.

Or the `k_*` APIs need to clearly describe which of them are safe to invoke from interrupt context or when a spinlock is held by the caller.  (Which, if it isn't either "all of them" or "none of them" should probably be done anyway.)

pizi-nordic, 2019-05-23:
Just for reference: Here is the list of affected APIs (calling z_reschedule*()):

- k_mbox_put()
- k_mbox_get()
- k_mbox_data_get()
- k_mbox_data_block_get()
- k_mem_slab_free()
- k_mem_pool_free()
- k_msgq_put()
- k_msgq_get()
- k_msgq_purge()
- k_pipe_put()
- k_poll_signal_raise()
- k_queue_cancel_wait()
- k_queue_insert()
- k_queue_append()
- k_queue_prepend()
- k_queue_alloc_append()
- k_queue_append_list()
- k_sem_give()
- k_stack_push()
- k_thread_start()
- k_thread_suspend()
- k_thread_resume()
- k_thread_abort()
- k_wakeup()
- k_timer_stop()
- k_thread_priority_set()

andyross, 2019-05-23:
I can do this, yeah.  I know enough about x86, ARM and Xtensa to do it immediately.  RISC-V and ARC I'll need to study a bit, and honestly I have no idea what the masking mechanism looks like on Nios II.  The issue of what to do with respect to the "locked" reschedule variants is still an open question though.   

The callers so far have always been able to expect that they'll be atomically swapped out, and if we break that we can do things like create deadlocks.  It's a change in the promise of the API that's going to take some auditing.  Some of these usages just may never work in an atomic context.

nordic-krch, 2019-05-24:
#16194 will be blocked until this one is fixed. There adding following change to logging log message which can happen from any context:
```
int err = k_mem_slab_alloc(&log_msg_pool, (void **)&msg, !k_is_in_isr() ? TIMEOUT : K_NO_WAIT);
```
and ideally i should do something like:
```
int err = k_mem_slab_alloc(&log_msg_pool, (void **)&msg, !k_is_in_atomic() ? TIMEOUT : K_NO_WAIT);
```

commit 0bf3a62a (2019-06-03): kernel/sched: Don't reschedule inside a nested lock

The internal "reschedule" API has always understood the idea that it
might run in a ISR context where it can't swap.  But it has always
been happy to swap away when in thread mode, even when the environment
contains an outer lock that would NOT be expecting to swap!  As it
happened, the way irq locks are implemented (they store flag state
that can be restored without context) this would "work" even though it
was completely breaking the synchronization promise made by the outer
lock.

But now, with spinlocks, the error gets detected (albeit i [... cut]
commit b3533a2d (2019-06-03): kernel/sched: Don't reschedule inside a nested lock

The internal "reschedule" API has always understood the idea that it
might run in a ISR context where it can't swap.  But it has always
been happy to swap away when in thread mode, even when the environment
contains an outer lock that would NOT be expecting to swap!  As it
happened, the way irq locks are implemented (they store flag state
that can be restored without context) this would "work" even though it
was completely breaking the synchronization promise made by the outer
lock.

But now, with spinlocks, the error gets detected (albeit i [... cut]
commit 312b43f1 (2019-06-03): kernel/sched: Don't reschedule inside a nested lock

The internal "reschedule" API has always understood the idea that it
might run in a ISR context where it can't swap.  But it has always
been happy to swap away when in thread mode, even when the environment
contains an outer lock that would NOT be expecting to swap!  As it
happened, the way irq locks are implemented (they store flag state
that can be restored without context) this would "work" even though it
was completely breaking the synchronization promise made by the outer
lock.

But now, with spinlocks, the error gets detected (albeit i [... cut]
closed 2019-06-03 by andrewboie