zephyrproject-rtos/zephyr #16275: setting_init crashes on qemu_x64 when setting BT_SETTINGS

Vudentz, 2019-05-21:
Steps to reproduce the behavior:
build test/bluetooth/shell
make run

uart:~$ bt init

***** Kernel Panic! *****
Current thread ID = 0x004000e0 (shell_uart)
eax: 0xffffffea, ebx: 0x00000000, ecx: 0x00516000, edx: 0x00516014
esi: 0x0001fba3, edi: 0x00400414, ebp: 0x0050660c, esp: 0x005065fc
eflags: 0x00000282 cs: 0x0008
call trace:
eip: 0x0000a509
     0x0000a546 (0x506620)
     0x0000e3f3 (0x5161b4)
     0x00011af6 (0x28760)
     0x0001f59d (0x1fba3)
     0x000063f6 (0x28760)
     0x0000659a (0xbc23)
     0x00005e3a (0x28760)
     0x000071e9 (0x0)
Terminate emulator due to fatal kernel error

addr2line -e zephyr/zephyr.elf 0x0009
/home/vudentz/git/zephyr-github/subsys/settings/src/settings_init.c:107

Disabling BT_SETTINGS makes it run properly.

rakons, 2019-05-27:
It seems that I have located the exact spot where the error appears. The fcb tries to read 2 bytes, but it hits this check inside flash_sim_read:

```
	if ((offset % FLASH_SIMULATOR_PROG_UNIT) ||
	    (len % FLASH_SIMULATOR_PROG_UNIT)) {
		return -EINVAL;
	}
```

Witch basically forbids to read unaligned data - the question is why. I am going to discuss it with the author and then we may offer some solution.

nvlsianpu, 2019-05-27:
^^ Can we make this configurable. Most of flash driver we have are able to unaligned read. I think only a few requires aligned start offset of readout, but still not aligned readout length.

nvlsianpu, 2019-08-12:
Was fixed by #16405

closed 2019-08-12 by nvlsianpu