zephyrproject-rtos/zephyr #16289: Driver: modem ublox-sara-r4 not compiling

WilliamGFish, 2019-05-21:
Attempting to use the UBlox srar-r4 modem driver and unable to compile:
DT_ not defined

**To Reproduce**
Added conf:
CONFIG_MODEM=y
CONFIG_MODEM_SHELL=y
CONFIG_MODEM_UBLOX_SARA_R4=y
CONFIG_NETWORKING=y
CONFIG_NET_OFFLOAD=y

# IPv6 support
CONFIG_NET_IPV6=n
CONFIG_NET_CONFIG_NEED_IPV6=n
# IPv4 support
CONFIG_NET_IPV4=y
CONFIG_NET_TCP=y
CONFIG_NET_UDP=y

1. west build -b particle_boron
2. See error

**Expected behavior**
Compile without error.

**Screenshots or console output**
```
_/drivers/modem/ublox-sara-r4.c:69:12: error: 'DT_UBLOX_SARA_R4_0_MDM_POWER_GPIOS_GPIO_CONTROLLER' undeclared here (not in a function); did you mean 'DT_JEDEC_SPI_NOR_0_WP_GPIOS_CONTROLLER'?
  PINCONFIG(DT_UBLOX_SARA_R4_0_MDM_POWER_GPIOS_GPIO_CONTROLLER,_
```

**Environment (please complete the following information):**
 - OS: Windows10
 - GCC
 - 1.14.99 Version used (zephyr-v1.14.0-930-g56e50cf43e23)

**Additional context**
I have other modem drivers working with similar set-up

WilliamGFish, 2019-05-21:
#14057 
The dts for the particle boron board mis-configured: (boards/arm/particle_boron/particle_boron.dts)

Currently is:
```
	sara_r4 {
		compatible = "ublox,sara-r4";
		label = "ublox-sara-r4";
		status = "ok";

		mdm-power-gpios-gpio = <&gpio0 16 0>;
		mdm-reset-gpios-gpio = <&gpio0 12 0>;
	};

```
Needs to have "-gpio" removed; see below

```
	sara_r4 {
		compatible = "ublox,sara-r4";
		label = "ublox-sara-r4";
		status = "ok";

		mdm-power-gpios = <&gpio0 16 0>;
		mdm-reset-gpios = <&gpio0 12 0>;
	};

```

Billy..

mike-scott, 2019-05-21:
@WilliamGFish Thanks for this!  I'll submit a fix today.

closed 2019-05-22 by nashif