zephyrproject-rtos/zephyr #16303: mbedtls: config-tls-generic.h: MBEDTLS_PLATFORM_NO_STD_FUNCTIONS seems ungrounded

pfalcon, 2019-05-21:
I recently spent a bunch of time debugging following issues: with debug logging enabled in mbedTLS, some messages are printed ok, some not, a bit of garbage was printed. Turned out, that we have MBEDTLS_PLATFORM_NO_STD_FUNCTIONS and MBEDTLS_PLATFORM_SNPRINTF_ALT defined in config-tls-generic.h. This causes mbedTLS to provide a dummy snprintf implementation, which then used in debug output which is broken.

However, mbedTLS isn't consistent with enforcing MBEDTLS_PLATFORM_NO_STD_FUNCTIONS in this way, and in other output just uses vsnprintf() without extra indirections similar to MBEDTLS_PLATFORM_SNPRINTF_ALT. Such output worked.

This makes me wonder why MBEDTLS_PLATFORM_NO_STD_FUNCTIONS is unconditionally defined at all. I guess (well, just guess so far) that the core of mbedTLS doesn't depend on too much of libc functionality. Then it's used only for optional things, like debug logging, and if user has enabled it, it would rather work out of box (by pulling in libc funcs, yeah, that's ok and expected).

Failing "work out of the box approach" (we shouldn't fail that), another approach would be "make configurable, and error out when needed, but not enabled". But the case "silently misbehave", as described above, should be avoided.

pfalcon, 2019-05-21:
To elaborate, the patch I'm looking forward to is:

~~~
-#define MBEDTLS_PLATFORM_NO_STD_FUNCTIONS
+//#define MBEDTLS_PLATFORM_NO_STD_FUNCTIONS
 #define MBEDTLS_PLATFORM_EXIT_ALT
 #define MBEDTLS_NO_PLATFORM_ENTROPY
 #define MBEDTLS_NO_DEFAULT_ENTROPY_SOURCES
-#define MBEDTLS_PLATFORM_PRINTF_ALT
-#define MBEDTLS_PLATFORM_SNPRINTF_ALT
+//#define MBEDTLS_PLATFORM_PRINTF_ALT
+//#define MBEDTLS_PLATFORM_SNPRINTF_ALT
~~~

Before submitting that, I want to discuss any possible reason why MBEDTLS_PLATFORM_NO_STD_FUNCTIONS was enabled.

pfalcon, 2019-05-21:
See also https://github.com/zephyrproject-rtos/zephyr/pull/16304 . While this ticket says: "we should not forcibly disable and mock libc functions in mbedtls, let them be", #16304 goes further, and says: "if user enabled extended API support in Zephyr config, automagically use those APIs with mbedTLS".

rlubos, 2019-05-22:
The answer for "why" is simple, just as any other config that was originally present int there - it is included in https://github.com/zephyrproject-rtos/zephyr/blob/master/ext/lib/crypto/mbedtls/configs/config-mini-tls1_2.h, which `config-tls-generic.h` is based on. Why was it in there, I have no clue. At the point it was created the debug logs were just fine. But indeed, something got broken over time (mbedTLS update perhaps?), I've recently observed messy logs as well.

So from my side, if the change proposed fixes the issue, feel free to submit it.

pfalcon, 2019-05-22:
@rlubos, Ok, thanks.

pfalcon, 2019-05-22:
Posted https://github.com/zephyrproject-rtos/zephyr/pull/16315.

PR #16315 (2019-05-23): ext: mbedtls: Don't define MBEDTLS_PLATFORM_NO_STD_FUNCTIONS
That means that when mbedTLS will need functions like snprintf(),
printf(), a few others, they will be used from libc, as done
normally by most of software. Note that mentioned functions
aren't normally used on code paths doing TLS communication per se.
Instead, they are used by debug logging, additional "print info"
style functions, etc.

Before, when MBEDTLS_PLATFORM_NO_STD_FUNCTIONS was defined, those
functions were stubbed out, so silently didn't work as expected
(a specific symptom seen was that some debug messages were printed
 [... cut]
commit f0ea13d2 (2019-05-23): ext: mbedtls: Don't define MBEDTLS_PLATFORM_NO_STD_FUNCTIONS

That means that when mbedTLS will need functions like snprintf(),
printf(), a few others, they will be used from libc, as done
normally by most of software. Note that mentioned functions
aren't normally used on code paths doing TLS communication per se.
Instead, they are used by debug logging, additional "print info"
style functions, etc.

Before, when MBEDTLS_PLATFORM_NO_STD_FUNCTIONS was defined, those
functions were stubbed out, so silently didn't work as expected
(a specific symptom seen was that some debug messages were printed [... cut]
closed 2019-05-23 by nashif