zephyrproject-rtos/zephyr #16313: LMP Response Timeout / LL Response Timeout (0x22) after ~40s when using LE Secure Connections

PhilippFinke, 2019-05-22:
**Describe the bug**
I'm using the hci_usb sample (Zephyr v1.14) on the nrf53840_pca10056 board as a BLE-USB-Dongle under Linux (bluez 5.50). I want to connect a custom peripheral running Zephyr v14.0 on a nrf52840. The peripherial uses a fixed passkey and issues bt_conn_security(conn, BT_SECURITY_FIPS) in its connected event to establish a LE Secure Connection. After entering the correct passkey the connection will be established and the devices will get paired.

Unfortunately the connection breaks down after around 40s with a LMP Response Timeout / LL Response Timeout (0x22).

I also tried a HAMA BT 4.0 USB-Dongle on the same Linux machine and got no problem connecting and pairing the very same peripheral.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nrf52840_pca10056 samples/bluetooth/hci_usb
2. west flash
2. Connect nrf52840_pca10056 to a Linux machine (e.g. fedora)
3. On Linux:
    bluetoothctrl --agent KeyboardDisplay
    [bluetooth]# default-agent
    [bluetooth]# select (mac addr of hci_usb dongle)
    [bluetooth]# pairable on
    [bluetooth]# scan on
    [bluetooth]# scan off (after the desired peripheral has been found)
    [bluetooth]# pair (mac addr of peripheral)
    [bluetooth]# (enter passkey)
    [bluetooth]# trust (mac addr of peripheral)
    [bluetooth]# connect (mac addr of peripheral)
4. Running btmon simultaneously I get following output (I skipped the pairing part here, only the connection establishment is shown):

[zephyr_dongle.pdf](https://github.com/zephyrproject-rtos/zephyr/files/3206446/zephyr_dongle.pdf)

**Expected behavior**
As mentioned I tried the same with a HAMA BT 4.0 dongle. With that I get following output:

[hama_dongle.pdf](https://github.com/zephyrproject-rtos/zephyr/files/3206447/hama_dongle.pdf)

**Impact**
Showstopper

**Screenshots or console output**
Attached files zephyr_dongle.pdf and hama_dongle.pdf shows the btmon output when establishing a connection. zephyr_dongle.pdf shows the LMP timeout at the end of the document.

**Environment (please complete the following information):**
 - OS: Linux fedora
 - Toolchain: GNU gcc
 - Zepyhr v1.14 (cebe115) both for hci_usb and custom peripheral.

**Additional context**
Add any other context about the problem here.

PhilippFinke, 2019-05-27:
After working with the HAMA dongle, I faced the same issue. I found out, that I had a bug in my custom peripheral FW. There I called a function in the BT connected event, that somehow took too long to process. This caused the error. I think this issue can be closed.

cvinayak, 2019-05-27:
@PhilippFinke you can close it yourself, if no action required on the issue you faced.

closed 2019-05-27 by PhilippFinke