zephyrproject-rtos/zephyr #16316: ST modules organization

erwango, 2019-05-22:
## Introduction

Due to move of `ext/hal/st` to modules a reorganization of ST libraries is needed.
This issue is mainly a formal request about what was discussed in #16044 and summarized in https://github.com/zephyrproject-rtos/zephyr/pull/16044#issuecomment-493023435

### Problem description

`ext/hal/st` contains 2 different types of 'libraries' (in the most generic acceptance of the word):
- Code that is supposed to run on STM32 products (STM32Cube HAL/LL, Product specific IPC library). This restriction does not come from license type but purely that it is made to drive STM32 HW.
- Code that is made to drive ST products, with no restriction on the CPU on which it could run (could be STM32 or also HW from any other vendor). This is typically small libraries driving a small ST component (sensor, RF module, ..)

Following introduction of Zephyr module, all code under `ext/hal/st` should be moved under zephyr module. Moving it as a whole inside a single ST module would imply that anyone that want to use a ST component on a non ST product would have to pull the whole STM32Cube packages (> 220MB).

### Proposed change

Split ST libraries into 2 modules:
- `hal_stm32`: For the code **meant to run** on STM32
- `hal_st`: For the code **meant to drive** ST components 

## Detailed RFC

Move code under `ext/hal/st` in 2 ST specific modules, using the following trees:

```
hal_stm32
 └── stm32cube
 │   ├── common
 │   ├── stm32f1
 │   ├── ...
 │   └── stm32wb
 └── lib
     ├── stm32wb
     │   └── hci  (cf #14188)
     └── ...

hal_st
 └── stsensors
 │   ├── audio
 │   │   └── microphone
 │   ├─ ....
 │   ├── proximity
 │   │   └── vl53l0x
 │   └── stmemsc  (cf #14111)
 │       ├── stmems_pids
 │       ├── a3g4250d
 │       ├── ais328dq
 │       └── ...
 └── strf
     ├── subg1 (cf #16086)
     └── ...
```

## Alternatives

Move `ext/hal/st` in a single ST module, with the drawbacks listed above.

erwango, 2019-05-22:
@avisconti, can you review ?

erwango, 2019-05-22:
@nashif, created `modules` label. Feel free to edit.

avisconti, 2019-05-22:
@erwango 
Yes, it seems aligned to what we said earlier.
Two points just for the sake of a better undertsanding:

1. Question: for microphone directory should we want to go with
            hal_st/stsensors/audio/microphone
    or with
            hal_st/stsensors/microphone

    I'm not against your proposal (first option), but I'm not sure we will ever have other than
    microphones sensors in audio.

2.  stmemsc and stmems_pids are basically the same thing.
     The first directory will be a3g4250d. But I guess it was just a typo.

erwango, 2019-05-22:
@erwango
> I'm not against your proposal (first option), but I'm not sure we will ever have other than microphones sensors in audio.

Ok, I guess that this is something that can be fixed on the way.

>  stmemsc and stmems_pids are basically the same thing.
>  The first directory will be a3g4250d. But I guess it was just a typo.

Updated

avisconti, 2019-05-22:
> @erwango
> 
> > I'm not against your proposal (first option), but I'm not sure we will ever have other than microphones sensors in audio.
> 
> Ok, I guess that this is something that can be fixed on the way.
> 

Yes, sure

> > stmemsc and stmems_pids are basically the same thing.
> > The first directory will be a3g4250d. But I guess it was just a typo.
> 
> Updated

Are you going to prepare the repos or will be @nashif doing that?

erwango, 2019-05-22:
> Are you going to prepare the repos or will be @nashif doing that?

I'm following recent rules concerning modules (https://github.com/zephyrproject-rtos/zephyr/pull/16081).

PR #16892 (2019-06-18): ext: hal: st: move to modules
split ext/hal/st into two modules:

- hal_stm32: For the code meant to run on STM32
- hal_st: For the code meant to drive ST components

Fixes #16316
commit faa398f0 (2019-06-18): ext: hal: st: move to modules

split ext/hal/st into two modules:

- hal_stm32: For the code meant to run on STM32
- hal_st: For the code meant to drive ST components

Fixes #16316
closed 2019-06-18 by nashif