zephyrproject-rtos/zephyr #16318: net: Network Offloading: Particle Boron

WilliamGFish, 2019-05-22:
**Describe the bug**
When attempting to use Offloaded Modem driver keep receiving MPU fault, see below.

The same driver works fine in v1.14.00; yet consistently fails in v1.14.99

**Expected behavior**
The driver should had over to the kernel once completed connection, as in previous version of Zephyr RTOS

**Impact**
Showstopper, using the same framework to build further modem drivers

**Screenshots or console output**
Once offloaded driver completes booting receive the following error:
```
Initialising Modem....
RSSI: -71 
***** MPU FAULT *****
  Instruction Access Violation
***** Hardware exception *****
Current thread ID = 0x200048e0
Faulting instruction address = 0x200048e0
Fatal fault in essential thread! Spinning...
```

**Environment (please complete the following information):**
 - Windows 10
 - GCC
 - Version 1.14.99

mike-scott, 2019-05-23:
This is for Boron 2G/3G w/ SARA modem driver?

WilliamGFish, 2019-05-23:
Yes, very much based on the LTE version. Works ok(ish) with 1.14.00 with some reconnect issues which is partly due to keepalive options and using a cheap roaming sim to do testing.

It appears to happen when the offload function hands back to the OS to boot.

weinholtendian, 2019-05-24:
I didn't see the exact same error, but I saw a stack check fail in main after enabling `CONFIG_MODEM_UBLOX_SARA_R4` in a new project. Could be the same issue? Setting `CONFIG_MAIN_STACK_SIZE=4096` helped.

WilliamGFish, 2019-05-24:
@weinholtendian 
Thanks but already have increased stack size in 1.14.00.

Oanerer, 2019-06-07:
I've only just recently started using Zephyr altogether. With my custom board file for the UBlox C030-R410M I've run into similar issues. One thing I noticed is that when calling 

``net_if_up(ictx.iface);`` 

at the end of modem_reset(), 'ictx.iface' is actually a null pointer. I'm not that up to speed yet with Zephyr, but I'm guessing that might be related?

mike-scott, 2019-07-31:
I will look at this today.

jukkar, 2019-08-01:
Please, only one assignee otherwise it is not clear who will start to look at it.

WilliamGFish, 2019-08-06:
> I will look at this today.

@mike-scott if you need more info let me know, as i'm now back at my desk "working"..

mike-scott, 2019-08-08:
@WilliamGFish Can you test https://github.com/zephyrproject-rtos/zephyr/pull/18103 and see if MPU fault still exists?

mike-scott, 2019-08-09:
@WilliamGFish I think this can be closed right?

WilliamGFish, 2019-08-09:
RESOLVED:
Configuration issue, settings clash;
CONFIG_HW_STACK_PROTECTION=y
and 
CONFIG_MODEM_SHELL=y

Additionally, not setting CONFIG_MODEM_SHELL=y will cause failures.

closed 2019-08-09 by WilliamGFish

mike-scott, 2019-08-09:
@WilliamGFish Ah, I'm sorry I thought you explained it was due to the sample you were using and not due to the modem.  If you feel this is still a modem-related issue please re-open.