zephyrproject-rtos/zephyr #16323: net: ipv6: tcp: unexpected reply to malformed HBH in TCP/IPv6 SYN

rmstoi, 2019-05-22:
If Zephyr receives TCP/IPv6 SYN with malformed Hop-by-Hop having wrong length, it replies with ICMPv6 Destination Unreachable (Port unreachable). Since the length of Hop-by-Hop option is wrong, Zephyr should instead send ICMP parameter problem message or silently drop the packet.

Note that it this packet is dropped for ICMPv6 and UDPv6.

[bad-hbho-len.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/3207309/bad-hbho-len.pcap.gz)

jukkar, 2019-08-15:
@rmstoi can you check this one, I fixed the TCP option calc len which hopefully fixes this.

rmstoi, 2019-08-16:
@jukkar I checked it still fails. The problem is that unknown option 0x42 in HBH has wrong length, and because of this packet must be dropped.

jukkar, 2019-08-26:
We can have this for 2.0 so adjusting target.

commit 0049c526 (2019-08-27): net: ipv6: Check sub-option length

Make sure that the extension, like HBHO, sub-option length is
not too large.

Fixes #16323
commit c73dae97 (2019-08-27): net: ipv6: Check sub-option length

Make sure that the extension, like HBHO, sub-option length is
not too large.

Fixes #16323
commit 50b6e3fa (2019-08-27): net: ipv6: Check sub-option length

Make sure that the extension, like HBHO, sub-option length is
not too large.

Fixes #16323
PR #18650 (2019-08-27): net: ipv6: Check sub-option length
Make sure that the extension, like HBHO, sub-option length is
not too large.

Fixes #16323
closed 2019-08-27 by ioannisg