zephyrproject-rtos/zephyr #16327: doc: networking: overview has out of date info for LwM2M

mike-scott, 2019-05-22:
**Describe the bug**
The networking overview page: https://docs.zephyrproject.org/latest/guides/networking/overview.html states that LwM2M doesn't support DTLS or Bootstrap mode.  This was added a while back and doesn't match with the rest of the docs.

**To Reproduce**
View: https://docs.zephyrproject.org/latest/guides/networking/overview.html

**Expected behavior**
LwM2M documentation should match throughout the docs.

**Impact**
Developers are getting misinformation and possibly avoiding Zephyr use.

ioannisg, 2019-09-03:
Are you going to address this for 2.0 @mike-scott ?

mike-scott, 2019-09-03:
Wow.. this totally feel off my plate.  Let me look.

closed 2019-09-04 by ioannisg