zephyrproject-rtos/zephyr #16330: LPCXpresso55S69 secure/non-secure peripheral split

agansari, 2019-05-22:
we need to look at structuring this similar to how dts/arm/nordic/nrf9160* is done.  So that secure/non-secure can work cleanly in the future.

_Originally posted by @galak in https://github.com/zephyrproject-rtos/zephyr/pull/15647_

agansari, 2019-10-24:
@MaureenHelm @galak I'm currently working on this issue and it's still under development. I want to make a proposition regarding this SoC.

Have only 2 targets:
 - lpcxpresso55s69 - cpu0 S and cpu1 NS
 - lpcxpresso55s69_ns - cpu0 NS

Currently we only have lpcxpresso55s69_cpu0 as a target, but I find cpu related split irelevant for this SoC; this is because the second cpu does not have security extensions and a bunch of other feature (fpu, mpu, etc). The second core is considered a slave core and only has a math accelerator and a cryptography accelerator as far as i can see. This second core can't run without the first core, i.e. when SoC is powered, core 0 is started and core 1 is halted expecting reset from core 0.

I understand that both Zephyr and SoC design is AMP and building will result in creating a binary for each core, but some additional scripting we will be able to generate a single multi-core image.

phonkane, 2019-12-22:
@agansari I am investigating running an app on LPCXpresso55S69/Zephyr/NS. I tried your approach (copied the board and devicetree sources under new names, then modified the new dtsi file addresses to nonsecure ones) and it looks promising. I wonder if you are making progress on this one - did not find anything in https://github.com/agansari/zephyr at least.

PR #22461 (2020-03-12): LPC55xxx secure/non-secure split
Adds NS definitions for CPU0 and CPU1.
Preparing platform for future development, dual core, mpu, etc.

Fixes #16330
closed 2020-03-12 by jhedberg