zephyrproject-rtos/zephyr #16341: Bluetooth: GATT Server failed to send Service Change Indication

tedd-an, 2019-05-23:
**Describe the bug**
GATT server failed to send Service Change Indication.
Ref: PTS GATT/SR/GAS/BV-01-C

**To Reproduce**
Steps to reproduce the behavior:
This was tested with PTS (Bluetooth SIG testing tool)

**Expected behavior**
GATT Server (IUT) should send Service Change indication to the client.

**Impact**
Test fail

**Environment (please complete the following information):**
Using AutoPTS on Linux
HW: reel_board

jhedberg, 2019-05-23:
Is this a recent regression, or something that's been there for a longer time? Should the fix be back ported to LTS?

closed 2019-05-23 by jhedberg

tedd-an, 2019-05-23:
Hi @jhedberg 
> Is this a recent regression, or something that's been there for a longer time? Should the fix be back ported to LTS?

The issue was introduced with e0f3ab6 and it was a part of regression list. But we missed it I guess. 
It doesn't need to be backported.

PR #16342 (2019-05-23): Bluetooth: GATT: Fix using a wrong handle
This patch fixes to use the right handle value.

Fixes #16341
commit fccb2e1f (2019-05-23): Bluetooth: GATT: Fix using a wrong handle

This patch fixes to use the right handle value.

Fixes #16341