zephyrproject-rtos/zephyr #16343: Provide functionality which allows checking if address is in read only memory

nordic-krch, 2019-05-23:
**Is your feature request related to a problem? Please describe.**
Logger needs to distinguish string addresses between read only locations and read write locations. 

**Describe the solution you'd like**
I would like to have public API call like `is_rodata(void *addr)`.

**Describe alternatives you've considered**
Alternatively, logger handles that locally but since it's generic functionality it would be better to have that in public API.
https://github.com/zephyrproject-rtos/zephyr/blob/8567bd6a75592727a6f43dbb1d27d7905324d800/subsys/logging/log_core.c#L125

carlescufi, 2019-05-23:
Why not send a PR that moves the implementation that you already have in `log_core.c` to a new file in `lib/`?

nordic-krch, 2019-05-23:
Because i'm not sure where to put it. those who has better system overview may have some suggestions.

carlescufi, 2019-05-23:
I see 2 ways to go about this:

1. Add a new arch-specific function `arch_is_rodata()` that each architecture fills in, and then have a common entry point `is_rodata()` that calls the arch-specific code. This common entry point could be in `linclude/kernel.h` or `include/misc/addr.h`
2. Add a new function that is identical to the current one in `log_core.c` in `lib/os/addr.c` and add a new `include/misc/addr.h`

ioannisg, 2019-05-23:
@nordic-krch what is the purpose of this? Could you elaborate more?

Btw, if you need to, generally, assess in run-time the access permissions of an address, you could, alternatively, implement this using the mpu.

nordic-krch, 2019-05-23:
@ioannisg i needed in the logger when i've added a functionality to detect logger misuse. Basically, in the logger you need to take special care if you are using %s with strings that can change between the moment log message is created and processing. I need to verify string address if and scream if it is not in read only memory. See #16310

tbursztyka, 2019-05-24:
That would be a nice addition (There are other place we could take advantage of such function).

However I would name the function sys_is_rodata() for instance, or if you make into kernel.h : k_is_rodata()

andrewboie, 2020-03-24:
Just have the logger check if the address is in the interval defined by the linker variables (_image_rodata_start, _image_rodata_end) defined in include/linker/linker-defs.h we don't need an API for this