zephyrproject-rtos/zephyr #16363: Error building x_nucleo_iks01a1 sample on nucleo_wb55rg after activating I2C Bus

rgf97, 2019-05-23:
**Describe the bug**
Hi, after making the configuration to activate I2C Bus on nucleo_wb55rg, i get this error while building x_nucleo_iks01a1 sample: `west build -b nucleo_wb55rg samples/shields/x_nucleo_iks01a1/`

`CMake Error at ../../../cmake/extensions.cmake:378 (add_library):
  No SOURCES given to target: drivers__i2c
Call Stack (most recent call first):
  ../../../cmake/extensions.cmake:355 (zephyr_library_named)
  ../../../drivers/i2c/CMakeLists.txt:3 (zephyr_library)
`

Any idea about this? Thx

rgf97, 2019-05-24:
Nvm

closed 2019-05-24 by rgf97

erwango, 2019-11-21:
For an unknown reason, this issue appear in list of open low prio bug whil it is closed.
Removing bug labels..

Trubochkin, 2020-07-08:
I have same error (used nRF52840 and Zephyr v2.2.1):

> Executing task: west build -b nrf52840_ble_gsm_gw <

-- west build: build configuration:
       source directory: C:\Users\Oleksii\Work\VIRAVIX\PROG\Nordic\Projects\fw.ble-gsm-gateway.nrf52.sim800
       build directory: C:\Users\Oleksii\Work\VIRAVIX\PROG\Nordic\Projects\fw.ble-gsm-gateway.nrf52.sim800\build
       BOARD: nrf52840_ble_gsm_gw (origin: CMakeCache.txt)
-- west build: building application
[0/1] Re-running CMake...
-- Zephyr version: 2.2.1
-- Selected BOARD nrf52840_ble_gsm_gw
-- Found west: C:/Python37/Scripts/west.exe (found suitable version "0.7.2", minimum required is "0.6.0")
-- Loading C:/Users/Oleksii/Work/VIRAVIX/PROG/Nordic/Projects/fw.ble-gsm-gateway.nrf52.sim800/boards/arm/nrf52840_ble_gsm_gw/nrf52840_ble_gsm_gw.dts as base
Devicetree header saved to 'C:/Users/Oleksii/Work/VIRAVIX/PROG/Nordic/Projects/fw.ble-gsm-gateway.nrf52.sim800/build/zephyr/include/generated/devicetree_unfixed.h'
Parsing C:/Users/Oleksii/zephyrproject/zephyr/Kconfig
Loaded configuration 'C:/Users/Oleksii/Work/VIRAVIX/PROG/Nordic/Projects/fw.ble-gsm-gateway.nrf52.sim800/boards/arm/nrf52840_ble_gsm_gw/nrf52840_ble_gsm_gw_defconfig'
Merged configuration 'C:/Users/Oleksii/Work/VIRAVIX/PROG/Nordic/Projects/fw.ble-gsm-gateway.nrf52.sim800/prj.conf'
Configuration saved to 'C:/Users/Oleksii/Work/VIRAVIX/PROG/Nordic/Projects/fw.ble-gsm-gateway.nrf52.sim800/build/zephyr/.config'
Kconfig header saved to 'C:/Users/Oleksii/Work/VIRAVIX/PROG/Nordic/Projects/fw.ble-gsm-gateway.nrf52.sim800/build/zephyr/include/generated/autoconf.h'
-- Cache files will be written to: C:\Users\Oleksii\AppData\Local/.cache/zephyr
-- Configuring done
CMake Error at C:/Users/Oleksii/zephyrproject/zephyr/cmake/extensions.cmake:372 (add_library):
  No SOURCES given to target: drivers__i2c
Call Stack (most recent call first):
  C:/Users/Oleksii/zephyrproject/zephyr/cmake/extensions.cmake:349 (zephyr_library_named)
  C:/Users/Oleksii/zephyrproject/zephyr/drivers/i2c/CMakeLists.txt:3 (zephyr_library)

CMake Generate step failed.  Build files cannot be regenerated correctly.
FAILED: build.ninja
"C:\Program Files\CMake\bin\cmake.exe" -SC:\Users\Oleksii\Work\VIRAVIX\PROG\Nordic\Projects\fw.ble-gsm-gateway.nrf52.sim800 -BC:\Users\Oleksii\Work\VIRAVIX\PROG\Nordic\Projects\fw.ble-gsm-gateway.nrf52.sim800\build
ninja: error: rebuilding 'build.ninja': subcommand failed
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\Users\Oleksii\Work\VIRAVIX\PROG\Nordic\Projects\fw.ble-gsm-gateway.nrf52.sim800\build'
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.